ARM-software/CMSIS_5 #937: 【CMSIS-DAP】 I have a question about CMSIS-DAP DAP_ProcessCommand.

balanceTWK, 2020-06-01:
[image]

[image]

The MDK did not process the subsequent operations are it because the DAP_Info function did not return the correct value ?

balanceTWK, 2020-06-01:
I'm currently trying to port cmsis-dap to the USB protocol stack of rt-thread (RTOS).

JonatanAntoni, 2020-06-02:
Hi @balanceTWK,

Its hard to tell you details without knowing your environment. 
Did you take a close look into the documentation at https://arm-software.github.io/CMSIS_5/DAP?

Do you have a test environment with a functional DAP implementation for?

Cheers,
Jonatan

balanceTWK, 2020-06-02:
> Hi @balanceTWK,
> 
> Its hard to tell you details without knowing your environment.
> Did you take a close look into the documentation at https://arm-software.github.io/CMSIS_5/DAP?
> 
> Do you have a test environment with a functional DAP implementation for?
> 
> Cheers,
> Jonatan

@JonatanAntoni  Thank you very much for your reminding.

I have been looking at the official documentation, but today I found out that cmsis-dap provides a Validation\MDK5\test.bat to test the DAP function. 囧rz。 I have solved my previous problem and now I am stuck in SWD/JTAG Communication Failure。

[image]

[image]

[image]

[image]

Does this mean there is a problem with my SWD GPIO simulation?Or something else?

JonatanAntoni, 2020-06-02:
Hi @balanceTWK,

well, going to analyse the actual serial wire communication between your DAP controller and the target processor can hardly be done on this forum.

Let me reraise one of my questions: Do you have a running DAP firmware for your environment?

I'd rather avoid changing multiple aspects at once. If its your primary interest to port CMSIS-DAP to another USB-stack implementation, I recommend to use a board which already has running CMSIS-DAP support. This would make it easier to compare the "original debugger firmware" with your modified one.

The serial wire communication is often realized using bit-banging the GPIO pins. This requires a very exact timing in your software. Therefore the DAP examples contain some macros for accurate busy waiting by doing a pre-calculated number of NOPs. Please be aware that the overall timing is very sensitive to compiler optimization. The example should work with Arm Compiler on highest optimization, only. If the timing is not exact, you might get it basically working when selecting a lower speed in MDK.

Cheers,
Jonatan

balanceTWK, 2020-06-02:
@JonatanAntoni  Thank you very much. I will solve the problem according to your way of thinking.😉

closed 2020-06-02 by balanceTWK