ARM-software/CMSIS_5 #938: Why RTX5 remove the task_id in the TCB?

swordligit, 2020-06-01:
I am porting from RTX to RTX5, just before there is a variable
`task_id` in the `OS_TCB`,  and as an convenient TCB access, e.g,
used for trace info, it is easy to map task_id to an task.
As for RTX5,  it is removed. Yes, I can use the `thread_id`, but the
`thread_id` is just pointer of an TCB, and is not so readable.
So if there is good reason for removing it? and can it be added back?

JonatanAntoni, 2020-06-02:
Hi @swordligit,

the `thread_id` is CMSIS-RTOS2 is not necessarily the a pointer to the TCB, this is an implementation detail you should not rely on. If you are looking for a human readable identification you can assign a string name by providing a user defined `osThreadAttr_t` when calling `osThreadNew`, see https://arm-software.github.io/CMSIS_5/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html#structosThreadAttr__t.

Does this help you?

Cheers,
Jonatan

swordligit, 2020-06-03:
Thank you Jonatan,
The thread name make sense, but OS will generate a SVC to
get it, something I think it is kind of high cost, because I want to print
it just before every trace.

JonatanAntoni, 2020-06-03:
Hi @swordligit,

yes, reading the thread names recurrently might cause a lot of overhead.

Please consider that RTX5 has been designed with having MPU protection in mind. In such a scenario threads might be executed in unprivileged mode while control blocks are kept in privileged memory (in order to protect the control blocks from malicious changes). Accessing the data (such as the name attribute in TCB) requires privilege elevation which can only be achieved by an SVC call. 

If you want to write trace messages, I'd use the ThreadId and translate ThreadIds to human readable names during post processing the trace on the host side. Unfortunately, `osThreadGetId()` uses SVC as well, so you'd need to cache the id's somehow in the thread local memories to access them without overhead.

Does this make sense for you?

Cheers,
Jonatan

swordligit, 2020-06-04:
`osThreadGetId() uses SVC (to read the value from RTOS control block) as well, so you'd need to cache the id's somehow in the thread local memories to access them without overhead.`
That's make sense.
Thanks for your detailed explain， help me a lot.

closed 2020-06-04 by swordligit