ARM-software/CMSIS_5 #940: Accuracy drop in the new arm_softmax_q7()

majianjia, 2020-06-06:
Hi,
I recently update CMSIS-NN from v1.1 to v1.3 and I found this accuracy dropping. 
I ran one of my old models using both versions with the same 1000 images from MNIST test set. The top 1 accuracy are:
with v1.1  95.4%
with v1.3  82.7%

The MCU I use is STM32L476 with macros : ` ARM_NN_TRUNCATE,  ARM_MATH_CM4, __FPU_PRESENT=1`

I compared the output of every layer with the same input data.
All layers are outputting the same result between the 2 versions but only those softmax output different values. 

I checked the file history and there is quite a huge modification. I will further investigate the issue and update in here. 
Jianjia

christophe0606, 2020-06-08:
It needs to be reverted to the old version. Some attempts where made to accelerate it without keeping the bit exactness with the old version.

On some networks it does not make a difference.

But it is not validated on enough networks and softmax is not the most consuming part of a network. So, the best is to revert to previous version.

majianjia, 2020-06-08:
Thanks for confirming the issue. 
Great work.

closed 2020-06-08 by majianjia