ARM-software/CMSIS_5 #943: Correct IRQ file for Cortex-M7?

maxgerhardt, 2020-06-11:
For GCC, the list of [IRQ implementation files](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/RTOS2/RTX/Source/GCC) contains, as I understand it:

* `irq_armv8mbl.S` (for ARMv8-M base-line)
* `irq_armv8mml.S` (for ARMv8-M mainstream-line)
* `irq_ca.S` (ARM Cortex-A / ARMv7A)
* `irq_cm0.S` (ARM Cortex-M0 / ARMv7M)
* `irq_cm3.S` (ARM Cortex-M3 / ARMv7M)
* `irq_cm4f.S` ((ARM Cortex-M4 / ARMv7R)

The [description](http://www.keil.com/pack/doc/CMSIS/General/html/index.html)  of RTOS2 howered says

-- | -- | --
RTOS v1 | Cortex-M0/M0+/M3/M4/M7 | Common  API for real-time operating systems along with a reference  implementation based on RTX. It enables software components that can  work across multiple RTOS systems.
RTOS v2 | All Cortex-M, Cortex-A5/A7/A9 | Extends  CMSIS-RTOS v1 with Armv8-M support, dynamic object creation, provisions  for multi-core systems, binary compatible interface.

So which IRQ implementation file is used for the Cortex-M7? Or for that matter, Cortex-M33, Cortex-M23, Cortex-M0+? Am I right to use the Cortex-M4F file fort the Cortex-M7 (which does compile)?

RobertRostohar, 2020-06-12:
The file naming convention for `irq_cmx.S` is from a long time ago when only Cortex-M0/M3/M4 existed.

The following mappings apply:
-	`irq_cm0.S`: Cortex-M0, Cortex-M0+, Cortex-M1, SC000
-	`irq_cm3.S`: Cortex-M3, Cortex-M4 without FPU, Cortex-M7 without FPU, SC300
-	`irq_cm4f.S`: Cortex-M4 with FPU, Cortex-M7 with FPU
-	`irq_armv8bl.S`: Cortex-M23 (Armv8-M Baseline)
-	`irq_armv8ml.S`: Cortex-M33, Cortex-M35P, Cortex-M55 (Armv8-M Mainline)
-	`irq_ca.S`: Cortex-A5/A7/A9

flit, 2020-06-12:
I'm not a fan of wrapping entire source files with `#ifdef`… Imo, it's better to get the build correct.

However, I do agree the irq file names need to be updated. This has caused me several minutes of confusion every time I have to look at it again.

Suggestion:

- `irq_cm0.S` ⇒ `irq_armv6m.S`
- `irq_cm3.S` ⇒ `irq_armv7m_nofp.S`
- `irq_cm4f.S` ⇒ `irq_armv7m_fp.S`
- `irq_ca.S` ⇒ `irq_armv7a.S`

It would also be nice to merge the v7-M files into a single `irq_armv7m.S` that uses the `FPU_USED` equate to include FPU support (though I would prefer a preprocessor macro instead of equate). (And the preprocessor directives in these files should also be unindented for better clarity.)

The same changes need to be made for all 3 compilers.

I'll be happy to create a PR (at some point) including all of the above.

RobertRostohar, 2020-06-15:
Thank you Maximillian and Chris for your comments.

I also agree that the IRQ files should be eventually renamed in a similar way that Chris suggested.

I’m also not a fan of adding the `#ifdef` wrappers since it clutters the file and more importantly the files are already described in the CMSIS-Pack (ARM.CMSIS.pdsc) which should be used by the IDEs and build systems (ex: CMSIS-Build).

Actually all CMSIS components are described as components which are used by various toolchains.

For example the RTX IRQ source files are described in the following way:
```
<file category="source" name="CMSIS/RTOS2/RTX/Source/ARM/irq_cm0.s"  condition="CM0_ARMCC"/>
…
<file category="source" name="CMSIS/RTOS2/RTX/Source/ARM/irq_cm4f.s" condition="CM4_FP_ARMCC"/>
…
<file category="source" name="CMSIS/RTOS2/RTX/Source/GCC/irq_cm0.S"  condition="CM0_GCC"/>
```

Conditions filter the architecture and the compiler.

Similar also the RTX pre-built libraries are described.

Therefore in toolchains with CMSIS-Pack management the correct files are automatically selected and the actual filename gets less important.

But as said at the beginning, the files should be renamed to better match and not introduce confusion.

We will look into this in the near future together with other enhancements.

RobertRostohar, 2021-05-17:
IRQ module names now match the architectures as suggested.

JonatanAntoni, 2021-06-08:
@maxgerhardt, may I ask you to check the fix and close the issue, please?

maxgerhardt, 2021-06-08:
Looks much better now with the corrected comments in the assembly files. Closing.

closed 2021-06-08 by maxgerhardt