ARM-software/CMSIS_5 #948: Exporting a PyTorch model to CMSIS-NN

Neosar, 2020-06-23:
Hello everyone,

I am working on a project that involves spoken digit recognition using their spectrograms. My model is exactly the same as the CIFAR-10 example. However, accuracy on the microcontroller (STM32F407VG) is quite low (about 10-20%).  I think the problem is the order of weights. In PyTorch, weight dimensions of Conv2d are [out_channel, in_channel, kernel[0], kernel[1]] (NCHW). I know I should reorder this for CMSIS-NN, but I came up with several formats. For example, in tutorial https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/converting-a-neural-network-for-arm-cortex-m-with-cmsis-nn/single-page It says the order should be **{in channel, x kernel, y kernel, out channel}**. However, in https://github.com/ARM-software/CMSIS_5/issues/343#issuecomment-382267724, it is said that it should be **[out_channel, height, width, in_channel]**. I would like to know which one is correct. 

And also, during training, I scale the pixel values by dividing them into 4, and I write this tensor to the header file (after converting it to HWC). Since I preprocess image beforehand, I use the code below,
`for (int i=0;i<33*33*3; i++) {
	  img_buffer2[i] = (q7_t)image_data[i];
  }`
Is this the correct way to do it?

Thank you.

christophe0606, 2020-06-24:
If you look at the inner loop of the code, you'll see that the input is organized like in the tutorial.

`Im_in[(in_row * dim_im_in_x + in_col) * ch_im_in + l]`

l is in channel. So the memory order is "in channel" first.

For the output it is the opposite:

`Im_out[i + (j * dim_im_out_x + k) * ch_im_out]`

i is "out channel".

Now regarding the conversion. I don't know what the type of image_data is but I would be surprised if a cast to q7 was enough.

felix-johnny, 2020-09-28:
@Neosar I am closing this ticket for inactivity for a long time. I agree with Christophe's observation on the input tensor though. If there is anything new in the issue, please feel free to re-open and thanks for using CMSIS-NN!

closed 2020-09-28 by felix-johnny