ARM-software/CMSIS_5 #951: What is the performance of DSP comparing to compiler optimization?

tkhamvilai, 2020-06-24:
I've seen that DSP relies a lot on loop unrolling techniques for optimizing the code. However, the compiler, e.g. gcc -O3, is more or less able to do the loop unrolling as well. Therefore, I wonder if there's any significant improvement using DSP over a regular compiler optimization.

Another related question is why the unrolling factor of 4 is chosen?

Best regards,

christophe0606, 2020-06-29:
It is highly compiler dependent. And indeed compilers have improved a lot and are now able to get good performance with automatic unrolling (CMSIS-DSP is an old library).

We unfortunately don't have the time to recheck every function with the main compilers (gcc and AC6) to check if the manual unrolling is still useful or not.

Ideally, we should remove it when it is no more useful since it would simplify the code.

The unrolling factor 4 is a good trade-off regarding the number of available registers. A compiler would be able to choose a better trade-off on a case per case basis.

christophe0606, 2020-06-29:
Also, I should add that compilers are focusing on float. For fixed point arithmetic they have more difficulties to generate good code.

closed 2020-07-06 by christophe0606