ARM-software/CMSIS_5 #953: Can't Compile A Minimal Working Example with CMSIS-DSP

ShikharJ, 2020-06-26:
I was trying to follow the instructions given under `Using the Library` section [here](https://arm-software.github.io/CMSIS_5/DSP/html/index.html), by including `DSP/Include/arm_math.h`, linking with `libarm_cortexM4l_math.a` and using the following command:
```
gcc main.c -o main -L. -l:libarm_cortexM4l_math.a 
```
but I get the following output:
```
In file included from main.c:2:
../../Include/arm_math.h:390:10: fatal error: cmsis_compiler.h: No such file or directory
  390 | #include "cmsis_compiler.h"
      |          ^~~~~~~~~~~~~~~~~~
compilation terminated.

```

Can anyone help me regarding this please?

jkrech, 2020-06-26:
Assuming your current directory is CMSIS/DSP/Lib/GCC
1) I would suggest that you do
`#include "arm_math.h"`
2) and specify the include path at the command line
`gcc main.c -o main -I../../Include -L. -l:libarm_cortexM4l_math.a `
3) spcify the include path to CMSIS-Core
`gcc main.c -o main -I../../Include -I../../../Core/Include -L. -l:libarm_cortexM4l_math.a `

Note, that when this repository is delivered in a pack (archive), both DSP and Core include files are copied into the directory `CMSIS/Include` therefore in this case you only had to specify a single include directory.

Does that work for you?

ShikharJ, 2020-06-26:
@jkrech Thanks for the clarification. This works out, can I ask the reason behind this change? I downloaded the latest release (5.7.0), albeit as a `.zip`.

jkrech, 2020-06-26:
I hate to say this, but the reason is backward compatibility to CMSIS V4 and before. At that point we had fewer components and header files ended up in CMSIS/Include. I do agree that we shall change the documentation reading to set both include paths separately.

ShikharJ, 2020-06-27:
@jkrech I changed the `main` to this:
```
#include <stdint.h>
#include "arm_math.h"

int main() {
	q15_t src[5] = {1, 2, 3, 4, 5};
	q15_t ret[5];
	q15_t scaleFract = 10;
	int8_t shift = 2;
	uint32_t blockSize = 5;
	arm_scale_q15(src, scaleFract, shift, ret, blockSize);
	return 0;
}
```
and got the following output:
```
gcc main.c -o main -I../../Include -I../../../Core/Include -L . -l:libarm_cortexM4l_math.a
/usr/bin/ld:./libarm_cortexM4l_math.a: file format not recognized; treating as linker script
/usr/bin/ld:./libarm_cortexM4l_math.a:1: syntax error
collect2: error: ld returned 1 exit status
```
I don't think this issue should arise, am I missing something here?

TTornblom, 2020-06-27:
Is “gcc” really your arm compiler? It is normally named something like “arm-none-eabi-gcc“

ShikharJ, 2020-06-27:
@TTornblom Ah, I should contextualize. I'm trying to get a script making use of `CMSIS` vector math operations running on an x86 to get a rough approximation of performance, before finally burning the code onto an ARM-Cortex M class device. So my motive here is to just get this code running on the host machine, and not the target board.

TTornblom, 2020-06-27:
But it appears you are trying to link against cortex m4 libraries, which won’t work.

ShikharJ, 2020-06-28:
I see. I believe the definition for the different functions are provided inside the `DSP/Source` folder. Is there any possible way of building and linking those files to run on the host machine?

TTornblom, 2020-06-28:
The sources for the DSP libraries are also in the pack. You should be able to either build the libraries for your architecture or just add the sources to the project and skip the libraries.

jkrech, 2020-06-28:
I think I need to set expectations. When we talk about gcc in the context of CMSIS we are referring to this variant of GNU GCC targeting 32-Bit Arm Cortex-M processors.
see: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm
It is a bit like trying to compile some windows software for Linux, there are some things that need explicit porting and may not work out of the box. The DSP library source code was specifically developed for Arm Cortex-M processors and not Intel x86.

ShikharJ, 2020-06-28:
Thanks @TTornblom @jkrech. I was able to get the code running on my host machine, and I'll probably spend the next few days burning it to my Cortex-M4 as well. I should close this issue.

closed 2020-06-28 by ShikharJ