ARM-software/CMSIS_5 #954: CMSIS Python wrapper example.py code. Test data link is dead, 404 error

PaulCreaserML, 2020-06-27:
Current link

https://www.physionet.org/pn3/ecgiddb/Person_87/rec_2.dat

Error 404

Perhaps the new link is:-

https://physionet.org/content/ecgiddb/1.0.0/Person_87/rec_2.dat

christophe0606, 2020-06-29:
Thank for reporting this.

christophe0606, 2020-06-29:
Corrected in commit 9aba007ce7f9faa29a1bd682dc8bf7f4db5b7fb4

closed 2020-06-29 by christophe0606

mrest-korg, 2021-03-11:
As a note: Public example did not work for me out of the box due to wrong scaling on the new test file: 

`# Data file from https://physionet.org/content/ecgiddb/1.0.0/Person_87/rec_2.dat`

Too high (> 1.0 gain), Q31 biquad example saturates.

christophe0606, 2021-03-11:
@mrest-korg I have just downloaded the updated pattern from 
https://archive.physionet.org/pn3/ecgiddb/Person_87/rec_2.dat

and I don't observe any saturation issues.

Can you open a new github issue if you also have problems with this pattern ?

mrest-korg, 2021-03-11:
@christophe0606  

This pattern works fine. No saturation, and also the original waveform as depicted in the various documentation sources.

They mus have changed the data and kept filenames the same. What a pity. 

Do you want me to open another issue and link your new file for you to create a pull request against?

christophe0606, 2021-03-11:
@mrest-korg If the new pattern is ok, I'll update the README and example with the new URL.
Thanks for reporting the problem.