ARM-software/CMSIS_5 #955: CMSIS Python wrapper example.py code.　Scipy and CMSIS results do not match

PaulCreaserML, 2020-06-28:
I went through the tutorial on the page below.

https://developer.arm.com/architectures/instruction-sets/dsp-extensions/dsp-for-cortex-m/how-to-use-the-python-wrapper-for-csmsis-dsp-with-biquads-as-an-example/single-page

The test data link is no longer alive, 404  error. I replaced the link with the below link.

https://physionet.org/content/ecgiddb/1.0.0/Person_87/rec_2.dat

The original data looks different. 

The results produced by the the scipy and cmsis-dsp filters look totally different.

- OS Ubuntu18.04
- Python 3.8
- Scipy 1.5
- Matplotlib 3.2.2
- CMSIS python setup.py build_ext --inplace ( GIT HEAD 614cb834a16dee6b25716a3fff2d2993e08152aa )

**Original data**
[image]

**Scipy Filter**

[image]

**CMSIS-DSP**

[image]

christophe0606, 2020-06-29:
I have just tested and see no difference between scipy and CMSIS-DSP (no visual difference because CMSIS-DSP example is in fixed point so will of course be a little bit different from scipy).

It looks like you have a problem reading the input file because mine is looking totally different.
The new link is pointing to the same file so your input should look the same as the one in the tutorial:

[image]

With the input you use it is normal scipy and CMSIS are giving different result : In Q31 format the input can't be bigger than 1 in absolute value so a rescaling is required.

PaulCreaserML, 2020-06-29:
My apologies. I actually made a mistake when downloading the test data in the command line (Corrected below)

wget -O rec_2.dat https://physionet.org/files/ecgiddb/1.0.0/Person_87/rec_2.dat?download

There example works perfectly

closed 2020-06-29 by PaulCreaserML