ARM-software/CMSIS_5 #956: [Build]  error: bad value (cortex-m7) for -mtune= switch

KyuhwanYeon, 2020-06-30:
Hello. I'm trying to build CMSIS/DSP

I followed [ReadMe](https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/DSP) in CMSIS/DSP

But  I failed to build it

OS: Ubuntu 16.04
Tool chain: GNU Arm Embedded Toolchain, Version 9-2019-q4-major

When I tried terminal
`cmake -DROOT="/home/kyuhwanyeon/workspace/2000_CMSIS_ws/CMSIS_5"   -DCMAKE_PREFIX_PATH="/home/kyuhwanyeon/gcc-arm-none-eabi-9-2019-q4-major/bin"   -DCMAKE_TOOLCHAIN_FILE="/home/kyuhwanyeon/workspace/2000_CMSIS_ws/CMSIS_5/CMSIS/DSP/armac6.cmake"   -DARM_CPU="cortex-m7"   -G "Unix Makefiles" ..`

The result
`-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- broken
CMake Error at /usr/share/cmake-3.11/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/make" "cmTC_389df/fast"
    /usr/bin/make -f CMakeFiles/cmTC_389df.dir/build.make CMakeFiles/cmTC_389df.dir/build
    make[1]: Entering directory '/home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    Building C object CMakeFiles/cmTC_389df.dir/testCCompiler.c.o
    /usr/bin/cc   -mcpu=cortex-m7 --target=arm-arm-none-eabi    -o CMakeFiles/cmTC_389df.dir/testCCompiler.c.o   -c /home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeTmp/testCCompiler.c
    cc: warning: ‘-mcpu=’ is deprecated; use ‘-mtune=’ or ‘-march=’ instead
    cc1: warning: command line option ‘-ftarget=arm-arm-none-eabi’ is valid for Java but not for C
    /home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeTmp/testCCompiler.c:1:0: error: bad value (cortex-m7) for -mtune= switch
     #ifdef __cplusplus
     ^
    CMakeFiles/cmTC_389df.dir/build.make:65: recipe for target 'CMakeFiles/cmTC_389df.dir/testCCompiler.c.o' failed
    make[1]: *** [CMakeFiles/cmTC_389df.dir/testCCompiler.c.o] Error 1
    make[1]: Leaving directory '/home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeTmp'
    Makefile:126: recipe for target 'cmTC_389df/fast' failed
    make: *** [cmTC_389df/fast] Error 2
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
See also "/home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeOutput.log".
See also "/home/kyuhwanyeon/workspace/2000_CMSIS_ws/BuildCMSISOnly/build/CMakeFiles/CMakeError.log".`

Could you explain what happend, and how can i solve this error?
Thank you

JonatanAntoni, 2020-06-30:
Hi @KyuhwanYeon,

I am not an expert in the CMake area. But it looks like in your environment the wrong compiler command is picked up, i.e. /usr/bin/cc probably doesn't point to the desired GNU Arm Embedded Toolchain. Can you please double check that?

Perhaps @christophe0606 can give you more details about CMake stuff.

Cheers,
Jonatan

christophe0606, 2020-06-30:
Hi @KyuhwanYeon,

For CMAKE_PREFIX_PATH, there should not be a bin at the end of the path

And since you are building with gcc, then the toolchain CMAKE_TOOLCHAIN_FILE should not point to armac6.cmake but to gcc.cmake

christophe0606, 2020-06-30:
I advise if you change those settings to clean your build folder before trying again : CMakeCache.txt, CMakeFiles ...
Otherwise you may have remaining issues

KyuhwanYeon, 2020-07-01:
Yeap. Now, I removed build folder and change CMAKE_PREFIX_PATH and CMAKE_TOOLCHAIN_FILE by your comment.

But result show another error
```
kyuhwanyeon@kyuhwanyeon-MS-7B17:~/workspace/2000_CMSIS_ws/BuildCMSISOnly/build$ cmake -DROOT="/home/kyuhwanyeon/workspace/2000_CMSIS_ws/CMSIS_5"   -DCMAKE_PREFIX_PATH="/home/kyuhwanyeon/gcc-arm-none-eabi-9-2020-q2-update/"   -DCMAKE_TOOLCHAIN_FILE="/home/kyuhwanyeon/workspace/2000_CMSIS_ws/CMSIS_5/CMSIS/DSP/gcc.cmake"   -DARM_CPU="cortex-m7"   -G "Unix Makefiles" .. 
CMake Error at /home/kyuhwanyeon/workspace/2000_CMSIS_ws/CMSIS_5/CMSIS/DSP/gcc.cmake:61 (add_link_options):
  Unknown CMake command "add_link_options".
Call Stack (most recent call first):
  /usr/share/cmake-3.11/Modules/CMakeDetermineSystem.cmake:94 (include)
  CMakeLists.txt:4 (project)

CMake Error: CMake was unable to find a build program corresponding to "Unix Makefiles".  CMAKE_MAKE_PROGRAM is not set.  You probably need to select a different build tool.
-- Configuring incomplete, errors occurred!

```

my cmake version is 11. and I also updated build-essentials.

christophe0606, 2020-07-01:
@KyuhwanYeon I am using cmake 3.14.3
I know there are issues with too recent ones. I don't know if 3.11 is ok. But it looks like it may be too old : "add_link_options" is not a recognized command.

Try with the 3.14.

TTornblom, 2020-07-01:
Looks like you don’t have a working “make” in your path.

Thomas

Thomas Törnblom, Product Engineer
IAR Systems AB
Box 23051, Strandbodgatan 1<x-apple-data-detectors://6/1>
SE-750 23 Uppsala, SWEDEN<x-apple-data-detectors://6/1>
Mobile: +46 76 180 17 80<tel:+46%2076%20180%2017%2080> Fax: +46 18 16 78 01<tel:+46%2018%2016%2078%2001>
E-mail: thomas.tornblom@iar.com<mailto:thomas.tornblom@iar.com> Website: www.iar.com<http://www.iar.com/>
Twitter: www.twitter.com/iarsystems<http://www.twitter.com/iarsystems>

KyuhwanYeon, 2020-07-06:
Thank you for your valuable comments. Now, I success to build
The problem was CMakeversion as @christophe0606  said.
I change the CMake version as 3.17. Then, I can build.

But I suggest modifying the CMakeList.txt

It said
cmake_minimum_required (VERSION 3.6)

But, It is not working as i mentioned in this issue.
I think minimum version should be changed as 3.14 

Thank you

christophe0606, 2020-07-06:
@KyuhwanYeon  I'll update the version number in cmake_minimum_required  in a future commit.

closed 2020-07-06 by christophe0606