ARM-software/CMSIS_5 #957: SVDConv and floating point (IEEE-754)

vitkorob, 2020-07-02:
Hello

Our team often creates IP cores that work with floating point (IEEE-754). It would be very convenient for us if SVDConv could support this type for generation. I suggest making support for float type.

I have tested the float type on SVDConv V3.3.35. An error has occurred.

```
*** ERROR M202: C:\projects\test\project\regmaps\test.svd (Line 744)
  Parse error: <dataType> = 'float'
```

Alignment and other generation rules should remain the same as for int32_t type. I think this feature can be pretty easy to implement.

renesas-brandon-hussey, 2021-05-20:
I have ran into this issue recently as well. If the SVDConv source was available then I could fix it and create a PR. This does not seem to be an option though according to https://github.com/ARM-software/CMSIS_5/issues/155. Is there any chance of this feature being added?

JonatanAntoni, 2021-05-25:
@thorstendb-ARM, can you comment on this, please?

JonatanAntoni, 2021-05-25:
Hi all, 

May I ask you to specify what exactly you would like to see SVDConv supporting? I.e. which data types shall be accepted? And where do you expect them to be defined in source code? E.g. shall we generate `#include "arm_math.h"` to pull in float data types like `float32_t` and `float64_t`?

Thanks,
Jonatan

renesas-brandon-hussey, 2021-05-25:
@JonatanAntoni I would like to see SVDConv be updated to support using `float` as an option for the `<dataType>` element. This would allow us to have float types in the resulting device header file. Currently in CMSIS-SVD.xsd the following options are available:

```xml
  <!-- dataType pre-defines the tokens in line with CMSIS data type definitions -->
  <xs:simpleType name="dataTypeType">
    <xs:restriction base="xs:token">
      <xs:enumeration value="uint8_t"/>
      <xs:enumeration value="uint16_t"/>
      <xs:enumeration value="uint32_t"/>
      <xs:enumeration value="uint64_t"/>
      <xs:enumeration value="int8_t"/>
      <xs:enumeration value="int16_t"/>
      <xs:enumeration value="int32_t"/>
      <xs:enumeration value="int64_t"/>
      <xs:enumeration value="uint8_t *"/>
      <xs:enumeration value="uint16_t *"/>
      <xs:enumeration value="uint32_t *"/>
      <xs:enumeration value="uint64_t *"/>
      <xs:enumeration value="int8_t *"/>
      <xs:enumeration value="int16_t *"/>
      <xs:enumeration value="int32_t *"/>
      <xs:enumeration value="int64_t *"/>
    </xs:restriction>
  </xs:simpleType>
```

If we use `<dataType>float</float>` in our SVD files then we get the M202 error.

> I.e. which data types shall be accepted?

`float` for IEEE 754 single-precision and `double` for IEEE 754 double-precision

> And where do you expect them to be defined in source code? E.g. shall we generate #include "arm_math.h" to pull in float data types like float32_t and float64_t?

What would be the benefit of using arm_math.h in this case? From a quick look at the file it appears to just use `float` and `double`. Could we use those types directly?

https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_math_types.h#L280

```c
  /**
   * @brief 32-bit floating-point type definition.
   */
  typedef float float32_t;

  /**
   * @brief 64-bit floating-point type definition.
   */
  typedef double float64_t;
```

JonatanAntoni, 2021-05-26:
Hi @renesas-brandon-hussey,

Using plain data types directly is often discouraged, e.g. by MISRA-C. Hence we are using typedefs indicating the actual data type size.

That's why I am a bit reluctant in simply adding `float` and `double`; and probably we should add `long double` aka `float128_t` and `float16_t` which has no ansi declaration, yet.

The values from SVD files are used directly for generating the header files. Hence we need to take care generated header files only make use of well known data types; and consider required includes.

Cheers,
Jonatan

renesas-brandon-hussey, 2021-05-26:
@JonatanAntoni I understand. I was looking to avoid bringing in more header files since we'll have to include `CMSIS_5/CMSIS/DSP/Include` now in addition to `CMSIS_5/CMSIS/Core/Include` for all projects. This is not something I'm overly concerned about though. Using `float32_t`, `float64_t`, and including `arm_math.h` is OK for my requirements.

thorstendb-ARM, 2021-05-27:
We should probably consider generating a union containing the float type and the corresponding uint.

JonatanAntoni, 2021-05-27:
@renesas-brandon-hussey, I think we should avoid requiring DSP/Include. This might need us to refactor the floating-point types into CMSIS-Core. We need to check if there is already some level of standard provided by compiler headers, such as ACLE. If not, we can add the types in a compiler specific way.

renesas-brandon-hussey, 2021-05-27:
@JonatanAntoni sounds good. Do you have any time frame for when this will be done?