ARM-software/CMSIS_5 #959: Cannot build CMSIS-NN for Mbed

dtch1997, 2020-07-07:
I am following the tutorial explained here: https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/single-page

The final step asks us to compile the image recognition app using: 
```
mbed compile -m DISCO_F746NG -t GCC_ARM --source .  --source ../ML-examples/cmsisnn-cifar10/code/m7 --source ../ML-examples/cmsisnn-cifar10/camera_demo/camera_app/ --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/Core/Include -j8 
```

However, this command fails with the error:
```
[Fatal Error] arm_bitonic_sort_f32.c@30,25: arm_sorting.h: No such file or directory
[ERROR] In file included from ../CMSIS_5/CMSIS/DSP/Source/SupportFunctions/SupportFunctions.c:30:0:
../CMSIS_5/CMSIS/DSP/Source/SupportFunctions/arm_bitonic_sort_f32.c:30:25: fatal error: arm_sorting.h: No such file or directory
 #include "arm_sorting.h"
                         ^
compilation terminated.
```
I am using CMSIS release 5.7.0.
```
$ git rev-parse HEAD
a65b7c9a3e6502127fdb80eb288d8cbdf251a6f4 
```

JonatanAntoni, 2020-07-07:
Hi @dtch1997,

since CMSIS 5.7.0 you need to add the `./CMSIS/DSP/PrivateInclude` of CMSIS-DSP to the compiler includes while compiling DSP source files. This happens automatically when using a CMSIS-Pack based build system. Otherwise you need to take care, manually.

Cheers,
Jonatan

dtch1997, 2020-07-07:
Thanks for the very prompt reply. I will attempt to recompile with the additional source library.

dtch1997, 2020-07-07:
The problem as originally described is resolved. I will close this issue.

closed 2020-07-07 by dtch1997