ARM-software/CMSIS_5 #960: Linker errors when compiling CMSIS-NN with Mbed

dtch1997, 2020-07-07:
I am following the tutorial explained here: https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/image-recognition-on-arm-cortex-m-with-cmsis-nn/single-page

Based on advice given in https://github.com/ARM-software/CMSIS_5/issues/959, my compile command is: 
```
mbed compile -m DISCO_F746NG -t GCC_ARM --source . --source ../CMSIS_5/CMSIS/NN/Include --source ../CMSIS_5/CMSIS/NN/Source --source ../CMSIS_5/CMSIS/DSP/Include --source ../CMSIS_5/CMSIS/DSP/Source --source ../CMSIS_5/CMSIS/Core/Include -j8 
```

The code compiles cleanly, but then I obtain several hundred linker errors of the form: 
```
BUILD/DISCO_F746NG/GCC_ARM/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q31.o: In function `arm_split_rfft_q31':
/home/dtch009/govtech/manucam/../CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_rfft_q31.c:243: multiple definition of `arm_split_rfft_q31'
BUILD/DISCO_F746NG/GCC_ARM/CMSIS_5/CMSIS/DSP/Source/TransformFunctions/TransformFunctions.o:/home/dtch009/govtech/manucam/../CMSIS_5/CMSIS/DSP/Source/TransformFunctions/arm_rfft_fast_init_f64.c:55: first defined here
```

How can I resolve this issue?

JonatanAntoni, 2020-07-07:
Hi @dtch1997,

I guess this happens due to the way mbed build system discovers source files to be included.

The DSP-Library consists of many medium sized source files, typically one per function. Compiling many files individually has a huge performance impact. Hence we added aggregator files which pull together all files per group. Using the aggregator file one only needs to compile one single file per group which appears to be much faster.

The downside is, when you blindly pick up all *.c files you'll end up with compiling both the aggregator *and* the individual files. Hence you will see multiple definitions of all functions.

As far as I understand the mbed build system correctly, you'd need to remove the aggregator files from your source tree to make this work. Same situation as in #959: Using a CMSIS-Pack compliant build system would do this automatically for you.

Cheers,
Jonatan

dtch1997, 2020-07-07:
@JonatanAntoni thanks once again for the prompt reply. I'm going to close the issue as the problem was resolved, but could you point me to some examples of CMSIS-Pack compliant build systems? The target architecture I am interested in is ARM Cortex M7, but my device is much lower-spec than a Raspberry Pi or Arduino.

closed 2020-07-07 by dtch1997

JonatanAntoni, 2020-07-07:
Hi @dtch1997,

there are commercial products using CMSIS-Packs, such as Arm Keil MDK or IAR Embedded Workbench. There is a plugin for GNU-ARM Eclipse available which is open source, see https://github.com/ARM-software/cmsis-pack-eclipse. And we are working on another command line based tool called CMSIS-Build, see  https://arm-software.github.io/CMSIS_5/Build/html/index.html.

[CMSIS-Pack](https://arm-software.github.io/CMSIS_5/Pack/html/index.html) is free and open for adoption by any other build system but its clearly up to the respective development team to device about integrating support or not. Unfortunately, no major open source build system has joined our efforts, yet.

Cheers,
Jonatan