ARM-software/CMSIS_5 #961: error: '__STATIC_FORCEINLINE' does not name a type; did you mean '__STATIC_INLINE'?

ardit93, 2020-07-07:
I'm creating a nn project on a nrf52 using a makefile on MacOS. At first I'm trying to reimplement the cifar example. I'm using CMSIS 5.6. I attached the makefile. I have already added all the needed source files in the makefile as well as the include folders. When I try to build the project, the following errors occur
/CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnsupportfunctions.h:210:1: error: '__STATIC_FORCEINLINE' does not name a type; did you mean '__STATIC_INLINE'?
 __STATIC_FORCEINLINE q31_t arm_nn_sat_doubling_high_mult(const q31_t m1, const q31_t m2)
 ^~~~~~~~~~~~~~~~~~~~
 __STATIC_INLINE
/CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnsupportfunctions.h:242:1: error: '__STATIC_FORCEINLINE' does not name a type; did you mean '__STATIC_INLINE'?
 __STATIC_FORCEINLINE q31_t arm_nn_divide_by_power_of_two(const q31_t dividend, const q31_t exponent)
 ^~~~~~~~~~~~~~~~~~~~
 __STATIC_INLINE

I tested the same configuration with CMSIS 5.4.0 and it compiled and linked successfully. The compiler, I'm using is gcc-arm 7.3.1.
[Makefile.txt](https://github.com/ARM-software/CMSIS_5/files/4887189/Makefile.txt)

JonatanAntoni, 2020-07-10:
Hi @ardit93,

it looks like your include folders are not set correctly.
Quickly glancing over your Makefile.txt:
```
  $(CMSIS_DIR)/CMSIS/Core/Include \
  $(CMSIS_DIR)/CMSIS/DSP/Include \
  $(CMSIS_DIR)/CMSIS/NN/Include \
  $(CMSIS_DIR)/CMSIS/Core_A/Include \
  $(CMSIS_DIR)/CMSIS/Driver/Include \
```
You must not add `Core/Include` and `Core_A/Include` at the same time. Both folders contain header files with the same name, hence you are causing ambiguities. If your target it Cortex-M then you only need `Core/Include`. 

I'd recommend to check the ARM.CMSIS.pdsc file to see which files and folders are required for your configuration. The define `__STATIC_FORCEINLINE` is in `Core/Include/cmsis_gcc.h`. Please be sure this header file is picked up, properly. You could for instance run the compiler with preserving the preprocessed source code (intermediate files) by adding `-save-temps` command line argument.

Cheers,
Jonatan

ardit93, 2020-07-10:
Hello Jonatan,

thanks for pointing that out. Although I removed the Core_A, the same error is coming up again.

JonatanAntoni, 2020-07-10:
Hi @ardit93,

please check in which compilation unit this error does occur, i.e. which `.c` file. Add `-save-temps` to the Compiler command line and attach the preprocessed intermediate file of this compilation unit. We need to check the actual preprocessing result and include chain. The should be a path ending up including `cmsis_gcc.h` file.

Cheers,
Jonatan

ardit93, 2020-07-10:
Hello Jonatan,

The error occurs in the arm_nnexamples_cifar10.cpp (provided by cmsis in nn examples)
In file included from /CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnfunctions.h:95:0,
                 from ../../../arm_nnexamples_cifar10.cpp:95:
/CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnsupportfunctions.h:210:1: error: '__STATIC_FORCEINLINE' does not name a ty

[arm_nnexamples_cifar10_output.zip](https://github.com/ARM-software/CMSIS_5/files/4903196/arm_nnexamples_cifar10_output.zip)

pe
 __STATIC_FORCEINLINE q31_t arm_nn_sat_doubling_high_mult(const q31_t m1, const q31_t m2)
 ^~~~~~~~~~~~~~~~~~~~
In file included from /CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnfunctions.h:95:0,
                 from ../../../arm_nnexamples_cifar10.cpp:95:
/CMSIS_5-5.6.0/CMSIS/NN/Include/arm_nnsupportfunctions.h:242:1: error: '__STATIC_FORCEINLINE' does not name a type
 __STATIC_FORCEINLINE q31_t arm_nn_divide_by_power_of_two(const q31_t dividend, const q31_t exponent)
 ^~~~~~~~~~~~~~~~~~~~
make: *** [_build/nrf52840_xxaa/arm_nnexamples_cifar10.cpp.o] Error 1

ardit93, 2020-07-10:
I saw that the makefile was including the cmsis_gcc.h included in the nordic sdk. Correcting the path resulted in another error(See the following errors). I could compile it by using the files provided by nordic but the copy_table variables are unknown to me. Coul you please explain to me what these variables are or direct me to the documentation resorces?

Thank you and cheers.
Ardit

In file included from /Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_compiler.h:54:0,
                 from /Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/DSP/Include/arm_math.h:310,
                 from ../../../arm_nnexamples_cifar10.cpp:91:
/Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_gcc.h: In function 'void __cmsis_start()':
/Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_gcc.h:146:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_start__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
   extern const __copy_table_t __copy_table_start__;
                               ^~~~~~~~~~~~~~~~~~~~
/Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_gcc.h:147:31: error: 'const __cmsis_start()::__copy_table_t __copy_table_end__', declared using local type 'const __cmsis_start()::__copy_table_t', is used but never defined [-fpermissive]
   extern const __copy_table_t __copy_table_end__;
                               ^~~~~~~~~~~~~~~~~~
/Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_gcc.h:148:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_start__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
   extern const __zero_table_t __zero_table_start__;
                               ^~~~~~~~~~~~~~~~~~~~
/Users/arditdvorani/Downloads/CMSIS_5-5.6.0/CMSIS/Core/Include/cmsis_gcc.h:149:31: error: 'const __cmsis_start()::__zero_table_t __zero_table_end__', declared using local type 'const __cmsis_start()::__zero_table_t', is used but never defined [-fpermissive]
   extern const __zero_table_t __zero_table_end__;
                               ^~~~~~~~~~~~~~~~~~

JonatanAntoni, 2020-07-10:
Hi @ardit93,

it looks like you are mixing up certain parts. If you are using the C startup routine from CMSIS you need to use the linker script and stuff as provided by CMSIS. 

The copy_table and zero_table symbols belong to the startup code required on bare metal targets. These linker generated tables specify which RAM areas needs to be initialized with static data or with zeros. I cannot give you much help about your specific application.

Please try to get a minimal example (i.e. empty main) running, first. If you are able to compile and run to main you are good to add more elaborated stuff.

Cheers,
Jonatan

closed 2020-07-14 by ardit93