ARM-software/CMSIS_5 #962: CMSIS-DAP: firmware version in DAP_Info

flit, 2020-07-09:
The DAP_Info command specification for the `DAP_ID_FW_VER` identifier simply states the value is "CMSIS-DAP Firmware Version". The reference CMSIS-DAP implementation currently returns either "1.2.0" or "2.0.0", depending on the `DAP_FW_V1` macro. This implies that `DAP_ID_FW_VER` is, in fact, the CMSIS-DAP protocol version. Another way to phrase this is that it's the version of the reference implementation upon which the firmware is built.

It would be very helpful to more clearly specify the format and usage of the `DAP_ID_FW_VER` value. In addition, we should add a value that is clearly specified as the "firmware implementation version" rather than protocol version.

The context here is the request from the microbit foundation in ARMmbed/DAPLink#723 to be able to access the DAPLink firmware version from the host (so they can inform users that they should update the firmware). The latest code in DAPLink reports its own version for `DAP_ID_FW_VER`. So instead of a value like "2.0.0", you'll get back something like "0254". That solves the request, but it's unfortunate that you cannot also determine the CMSIS-DAP protocol version, and it's not clear whether it's compliant with the protocol.

RobertRostohar, 2020-07-13:
You are right. The CMSIS-DAP Firmware Version as defined in the DAP_Info command could be phrased as the CMSIS-DAP protocol version.

The version is taken into account by the debuggers in order to identify features of CMSIS-DAP.

Revision history shows the versions and how they evolved:
http://www.keil.com/pack/doc/CMSIS/DAP/html/dap_revisionHistory.html

Reporting any other version would not be compliant and debuggers might not work correctly.

Firmware implementation versions and how they are reported are device specific. They could be reported through the CMSIS-DAP Vendor Commands which could be also used for firmware updates (not part of CMSIS-DAP).

flit, 2020-07-13:
Needing get the implementation version is a very common requirement. If for nothing else other than presenting to the user. So we should allocate a standard info identifier for the firmware version. What about ID 0x07 (just the next in sequence)? I would also rename the constants in the code to be more clear…

Proposal:
- Change 0x04
  - Constant in code: `DAP_ID_FW_VER` ⟹ `DAP_ID_CMSIS_DAP_VER`
    - Will keep `DAP_ID_FW_VER` available as an alias of `DAP_ID_CMSIS_DAP_VER`, but deprecated
  - Description in documentation: "CMSIS-DAP Firmware Version (string)" ⟹ "CMSIS-DAP Protocol Version (string), with a format like '2.0.0', nominally one of the versions from the CMSIS-DAP revision history."
- Add 0x07
  - `DAP_ID_PRODUCT_VER`
  - "Product Firmware Version (string), vendor-specified version format"

RobertRostohar, 2020-07-13:
Sounds like a good proposal. We will look into this in the near future.

microbit-carlos, 2021-01-20:
Are there any updates on this?

micro:bit V2 is in the process to be merged upstream on DAPLink, and it would be re really beneficial to be able to obtain the DAPLink version in our online editors.

flit, 2021-01-28:
@microbit-carlos The change is implemented in #973 but hasn't been merged yet.

thegecko, 2021-04-23:
@RobertRostohar , 0x07 is now in use by TARGET_BOARD_NAME for DAP_INFO. Can you recommend a different address?

We should also update the [web page](https://arm-software.github.io/CMSIS_5/DAP/html/group__DAP__Info.html) with TARGET_BOARD_NAME and TARGET_BOARD_VENDOR

flit, 2021-05-18:
I can change the ID and add TARGET_BOARD_NAME and TARGET_BOARD_VENDOR.

RobertRostohar, 2021-05-19:
We are going to update the documentation in the next days and address also the mentioned IDs.

RobertRostohar, 2021-06-01:
Product Firmware Version has been added as requested.

closed 2021-06-18 by flit