ARM-software/CMSIS_5 #966: AvgPool in a Q15_t model

dtch1997, 2020-07-15:
I am following [this tutorial](https://developer.arm.com/solutions/machine-learning-on-arm/developer-material/how-to-guides/converting-a-neural-network-for-arm-cortex-m-with-cmsis-nn/single-page) to convert my ONNX model to CMSIS-NN. 

I would like to have my whole model work in Q15. However I noticed there is no `arm_avgpool` that works with q15 input and no data conversion function to convert q15 to q7 to use `arm_avgpool_q7_HWC`. Does this mean I can't use AvgPool layers in the model?

christophe0606, 2020-07-16:
@dtch1997 In the CMSIS-DSP, you'll find a  arm_q15_to_q7 function
But you'll lose some accuracy by doing it so it is not the ideal solution.

dtch1997, 2020-07-20:
@christophe0606 thanks. It seems odd that there is no in-built q15 average pooling solution. Perhaps that could be a useful feature  in the future.

closed 2020-07-20 by dtch1997