ARM-software/CMSIS_5 #967: RTK Library and Microsemi SmartFusion Devices without LDREX/STREX.

dmg-dev, 2020-07-16:
Does the current CMSIS RTK (RTOS v2) library work with Microsemi SmartFusion devices that do not implement the LDREX and STREX instructions of the Cortex-M3?

### **Some background**
In the past, on older versions of the CMSIS RTK I had to build my own RTX_CM3X.lib library file with the 'CM3X_LE' Target Option.
See [http://www.keil.com/support/docs/3551.htm](http://www.keil.com/support/docs/3551.htm)

JonatanAntoni, 2020-07-16:
Hi @dmg-dev,

By default the Cortex-M3 build (Armv-7M) assumes availability of `LDREX` and `STREX` instructions for atomic memory accesses.

If your device does not have those you need to compile RTX5 from source code with having `EXCLUSIVE_ACCESS` defined to `0`. In this case atomic memory accesses are replaced by interrupt disable.

Cheers,
Jonatan

dmg-dev, 2020-07-16:
Thank you Jonatan.

closed 2020-07-16 by dmg-dev