ARM-software/CMSIS_5 #968: Q15 FFT performance improvement using GCC on Cortex-M4

rsaxvc, 2020-07-16:
I noticed that GCC(GNU Tools for ARM Embedded Processors), when targeting Cortex-M4 processors with DSP extensions, is about 4-5% slower than it could be, for the following reasons(when combined, come to around 4-5%, depending on FFT size):

- The Q15 FFTs are not hooked up to the CMSIS PKHBT intrinsics
- The CMSIS PKHBT intrinsics are not hooked up to the PKHBT instructions on GCC
- ARM Embedded Processors GCC -O3 defaults to 2-byte loop alignment, which can lead to a misaligned 4-byte instruction at the top of the loop, causing a delay/stall each time the loop is updated. This can be fixed with -falign-loops=4

A pull request for PKHBT is here:
https://github.com/ARM-software/CMSIS_5/pull/952

christophe0606, 2020-07-20:
Thanks for reporting this. I have no idea why the PKHBT were disabled a few years ago.

christophe0606, 2020-07-23:
I have merged your changes in commit 0ec39ae4faddaf186943d307f2420e60e72b924f and 8a1fd91319106663f74ca03353a26301f5512fca

closed 2020-07-23 by christophe0606