ARM-software/CMSIS_5 #969: Purpose of BufferA, BufferB in CMSIS-NN functions

dtch1997, 2020-07-20:
In many CMSIS-NN functions, for example `arm_convolve_HWC_q15_fast`, we have the option to provide an argument `bufferA` and `bufferB`. It is unclear to me what these are supposed to be. In some example applications I have seen them both set to NULL and in others I have seen them set to some value. How am I supposed to use these arguments?

majianjia, 2020-07-20:
@dtch1997 
There are some descriptions in their headers, including the size [example](https://github.com/ARM-software/CMSIS_5/blob/098cf8406f1463fc2cde3ee0ed8fecf4eba0690e/CMSIS/NN/Include/arm_nnfunctions.h#L784). 
They are mainly for the intermediate buffers. Such as im2col buffer in convolution. or line buffer in pooling. 

For the latest versions, you may call a [provided function](https://github.com/ARM-software/CMSIS_5/blob/098cf8406f1463fc2cde3ee0ed8fecf4eba0690e/CMSIS/NN/Include/arm_nnfunctions.h#L242) to get the size of the buffer.

dtch1997, 2020-07-21:
Thanks, that clears it up.

closed 2020-07-21 by dtch1997