ARM-software/CMSIS_5 #970: CMSIS Build makefile produces erroneous "CPU is not supported" for Cortex-M0+

bscarlet, 2020-07-23:
Using CMSIS Build with gcc to build for an Atmel/Microchip SAMD21G18A, the generated makefile, when used, produced the error "CPU is not supported."

The problem seems to be caused by inconsistent use of the make variable "CPU" - the value in the variable was "Cortex-M0+" but the value handled by the makefile (in my case GCC.9.2.1.mak, included by the generated makefile) is "Cortex-M0plus".

This patch adds support for the "Cortex-M0+" value, and fixed the problem for me.
[cmsis-build-support-m0+-patch.txt](https://github.com/ARM-software/CMSIS_5/files/4967154/cmsis-build-support-m0%2B-patch.txt)

Because the makefile generation code is closed source, I cannot trace the origin of the value in the variable to form an opinion as to which, if either, value is "right". However, it seems possible that the value comes from a field of the input .pdsc file, e.g. a value of a Dcore attribute. If that's the case, then - as both the chip-vendor (Microchip/Atmel) supplied .pdsc file and ARM.CMSIS.pdsc from this project both use "Cortex-M0+" - not only does supporting that value seem warranted, but perhaps the "Cortex-M0plus" value currently supported in GCC.9.2.1.mak is actually "wrong" in some sense.

Note also that the analogous makefiles for other toolchains seem to have similar code and so may have the same problem.

jkrech, 2020-07-23:
Well spotted. PACK.xsd specifies the strings in DcoreEnum hence Cortex-M0plus is in fact wrong and needs to be replaced in the toolchain specific *.mak files with Cortex-M0+
Thanks for reporting.

jkrech, 2020-10-02:
Fixed in CMSIS-Build 0.10.0 (https://github.com/ARM-software/CMSIS_5/releases/download/5.7.0/cbuild_install.0.10.0.sh)

closed 2020-10-02 by jkrech