ARM-software/CMSIS_5 #971: CMSIS Build makefile, with GCC, always rebuilds library

bscarlet, 2020-07-23:
For a cprj/target/output of type="lib", the generated makefile creates a makefile target named $(TARGET).lib, even though the rule for that target picks up the definition of AR_FLAGS from the toolchain-specific makefile, which - for gcc - creates lib$(TARGET).a instead.

As a result, when make is invoked to bring the library up to date, it cannot see the timestamp on lib$(TARGET).a and so understand that no work is necessary. Instead, it thinks it's looking for $(TARGET).lib, and because that file is never present it always re-runs the rule.

In my case, this means everything downstream in a larger build also gets rebuilt - i.e. the problem cascades.

I suggest that the generated makefile use the actual name of the library for the target.

If in the GCC .mak file there were
library_name:=lib$(1).a
and in the ARMCC .mak files there were
library_name:=$(1).lib

then the generated makefile could contain $(OBJ_DIR)/$(call library_name,$(TARGET)) where it currently contains $(OBJ_DIR)/$(TARGET).lib

and I think that would fix the problem.

brondani, 2020-07-24:
Hi Benjamin, thanks for reporting this issue.
Your proposed solution is also valid - it would just need backspaces' removal in order to support paths with (escaped) whitespaces.
We intend to release an updated beta version by mid of August, I will keep you posted!

jkrech, 2020-10-02:
Fixed in CMSIS-Build 0.10.0 (https://github.com/ARM-software/CMSIS_5/releases/download/5.7.0/cbuild_install.0.10.0.sh)

closed 2020-10-02 by jkrech