ARM-software/CMSIS_5 #974: Setting ARM_MATH_DSP and ARM_MATH_LOOPUNROLL on M3 Chip Increases Runtime

ShikharJ, 2020-07-28:
Hey everyone, I was experimenting a bit with my M3 board, and was trying to see if applying the ARM_MATH_DSP AND ARM_MATH_LOOPUNROLL compilation flags would lead to a performance change in the absence of a DSP accelerator chip. The runtime, for each of individual cases, as well as both flags being set, was worse than the runtime that I obtained without any flags being set. Is this expected behaviour?

christophe0606, 2020-07-29:
What compiler are you using ?

ShikharJ, 2020-07-29:
Compiler: `arm-none-eabi-gcc`
Compiler Options: 
```
set(COMMON_FLAGS "-mcpu=cortex-m3 -O3 -mthumb -Wall \
-fno-exceptions -ffunction-sections -fdata-sections -Wno-unused-variable \
-Wno-unused-parameter -Wno-parentheses \
-Wno-unused-function -ggdb -MD -fno-common -fmessage-length=0 -save-temps")
set(LINKER_FLAGS "-Wl,-gc-sections -mthumb --specs=nosys.specs ")
```
Matrix Type: `q15`

christophe0606, 2020-07-29:
In q15, CMSIS-DSP is using a function read_q15x2_ia to access memory.
In some cases, on gcc, this function is not optimized correctly.

With ARM_MATH_DSP defined, the arm_mat_mult_q15.c is using  read_q15x2_ia

So it may be the cause of the issue. Although, in general it occurs only when -fno-builtin is used and I don't see it in your compile options.

When I have more time, I'll try to reproduce the issue on my side.

What's your gcc version ?

ShikharJ, 2020-07-29:
Also, I think the issue is with the usage of `__SMLALD()` function for multiplication of those two variables obtained using `read_q15x2_ia`. It seems too expensive. My `gcc` version is `9.3.1`.

ShikharJ, 2020-07-30:
I get the following assembly for non-inlined `read_q15x2_ia`:
```
read_q15x2_increment:
.LVL35:
.LFB5:
	.loc 1 50 46 is_stmt 1 view -0
	.cfi_startproc
	@ args = 0, pretend = 0, frame = 0
	@ frame_needed = 0, uses_anonymous_args = 0
	@ link register save eliminated.
	.loc 1 51 3 view .LVU110
	.loc 1 50 46 is_stmt 0 view .LVU111
	mov	r3, r0
	.loc 1 51 11 view .LVU112
	ldr	r2, [r0]
	ldr	r0, [r2], #4	@ unaligned
.LVL36:
	.loc 1 52 3 is_stmt 1 view .LVU113
	.loc 1 52 9 is_stmt 0 view .LVU114
	str	r2, [r3]
	.loc 1 53 3 is_stmt 1 view .LVU115
	.loc 1 54 1 is_stmt 0 view .LVU116
	bx	lr
	.cfi_endproc
```
I believe, separately reading two `q15` from an array would take 4 cycles. This non-inlined version would take almost double of that.

christophe0606, 2020-07-31:
Normally gcc should be able to optimize out this read_q15x2_ia and inline and reduce it to the smallest number of instructions. I don't know why in your configuration it looks like it is not the case.

**If your data is aligned**, you may want to change the implementation of read_q15x2_ia  and reuse the old deprecated __SIMD32 macros.

I am unfortunately not able to re-check all of this on an M3 currently.

ShikharJ, 2020-07-31:
@christophe0606 Also, I have one other question regarding the use of a DSP chip. Does it make use of a different instruction set than the one on which a Cortex-M3 is based (pertaining to ARM V-7M architecture)? Or does the chip use the same generated assembly, but has the hardware specifically designed to run those assembly instructions like `read_q15x2_ia` as efficiently as possible?

christophe0606, 2020-07-31:
There is no DSP extension on Cortex-M3:

https://developer.arm.com/architectures/instruction-sets/dsp-extensions/dsp-for-cortex-m

So SMLALD should not be used.

gcc may have replaced it with something equivalent instead of just failing.

Some cortex-M have a DSP extensions (additional HW and instructions) to execute more efficiently some categories of code.

It is not the case of the Cortex-M3. And if ARM_MATH_DSP is not defined you should not define it. The definition is done automatically depending on the value of -mcpu and the recognized core.

So I think it explains your observation about the performance. And I believe we can close this issue.

closed 2020-07-31 by christophe0606