ARM-software/CMSIS_5 #980: CMSIS DSP:build error

swordligit, 2020-08-05:
Hi,
   I am updating the cmsis dsp lib to 1.8.0, according to readme, I can define "ARM_DSP_CONFIG_TABLES" and select special
table to save some code space.
  But when I won't select "ARM_TABLE_TWIDDLECOEF_F32_4096", there
will be a compile error:
```
||   CC      platform/cmsis/DSP_Lib/TransformFunctions/TransformFunctions.o
|| In file included from ../../platform/cmsis/DSP_Lib/TransformFunctions/TransformFunctions.c:40:
|| ../../platform/cmsis/DSP_Lib/TransformFunctions/arm_cfft_radix2_init_f32.c: In function 'arm_cfft_radix2_init_f32':
platform/cmsis/DSP_Lib/TransformFunctions/arm_cfft_radix2_init_f32.c|81 col 31| error: 'twiddleCoef' undeclared (first use in this function); did you mean 'twiddleCoef_64'?
||    81 |   S->pTwiddle = (float32_t *) twiddleCoef;
||       |                               ^~~~~~~~~~~
||       |                               twiddleCoef_64
```

So the problem is ' arm_cfft_radix2_init_f32' uses 'twiddleCoef' without condition.
I have removed the "arm_cfft_radix2_init_f32.c" in the 'TransformFunctions.c', and if
you have better advice, I will accept it.

christophe0606, 2020-08-06:
You're right. This function is deprecated. It has not been updated with the new flags for FFT tables but it was kept inside TransformFunctions.c to avoid breaking existing projects.

It is a bug. Either we need to add the flag or we need to remove it from TransformFunctions.c

I'll probably remove it from TransformFunctions.c

christophe0606, 2020-08-06:
I can't remove the radix4_init and it will cause the same issues. So I'll add compilation flags to avoid including the files in TransformFunctions.c if the related tables are not selected.

christophe0606, 2020-09-15:
I have corrected the issue in commit https://github.com/ARM-software/CMSIS_5/commit/0fb48db066cae491082b9ee52a870f79accd3a7a

swordligit, 2020-09-18:
OK. Thanks a lot!

closed 2020-09-18 by swordligit