ARM-software/CMSIS_5 #983: CMSIS-DSP: is it possible to use it for Cortex-A53

YuanfenZheng, 2020-08-12:
I have a quick question, is there a chance to use CMSIS-DSP for Cortex-A53?
Many thanks!

christophe0606, 2020-08-13:
We have made some preliminary work.

CMSIS-DSP has some Neon and in case of arm_dot_prod_f32.c even use a specific aarch64 intrinsics.

The only difficulty will be the folder Core_A in the CMSIS. It is not yet upgraded for aarch64

So I expect you'll have some build issues if you want to build for aarch64 but in principle there is no blocking point. Just some additional work remaining to do.

YuanfenZheng, 2020-08-13:
Thank you for your feedback, Christophe.
I am using Xilinx 2020.1 Vitis to build CMSIS-DSP object, and use it as a library for an application run on Cortex-A53.
1) Using Xilinx ARM v8 GNU Toolchain (aarch64-none-elf-gcc)
    -  Set Preprocessor Macros:  DISABLEFLOAT16, ARM_MATH_DSP, ARM_MATH_NEON, ARM_MATH_NEON_EXPERIMENTAL
    -  Build ARM Options:  mcpu=cortex-a53, march=armv8-a

    build fails with error of unknown mnemonic e.g. ssat.

2) Using Xilinx ARM v8 32 Bit GNU Toolchain (arm-none-eabi-gcc)
    -  Set Preprocessor Macros:  DISABLEFLOAT16, ARM_MATH_DSP, ARM_MATH_NEON, ARM_MATH_NEON_EXPERIMENTAL
    -  Build ARM Options:  mcpu=cortex-a53, mfpu=neon-vfpv4, mfloat-abi=hard

     build passes

It seems CMSIS-DSP is based on 32-bit instructions, therefore request a corresponding toolchain, and Cortex-A53 shall run in Aarch32 state?
If you see any of above setting is not proper, or you see a chance to make CMSIS-DSP run on Cortex-A53 in Aarch64 state at the moment, kindly drop me some hints what/where to modify.

The CMSIS source code I compile is:
  - dsp/source/*
  - dsp/computeLibrary/source/*

Corresponding header files(INC directory) are in
  - dsp/include
  - dsp/computeLibrary/include/*
  - dsp/privateInclude/*
  - core_A/include

christophe0606, 2020-08-13:
Indeed the problem is that CMSIS-DSP is using a few intrinsics which are defined in compiler files like Core_A/cmsis_gcc.h.

This file must be updated to also support aarch64.

closed 2020-08-13 by YuanfenZheng