ARM-software/CMSIS_5 #985: CMSIS NN Question - How to use activation function tanh?

wwilliams3111, 2020-08-14:
Hello, I've been trying to convert a simple NN model with 2 fc layers from MATLAB over to CMSIS NN implementtion. I have the weights properly quantized and am getting matching results from my MATLAB simulation and the output of my first fc layer in CMSIS NN when making a call using arm_fully_connected_q15( ). After this, I apply a tanh activation function, but the corresponding output from arm_nn_activations_direct_q15( ) gives me only the correct values for positive q15 numbers. All negative numbers in the q15 format are being zeroed out. I'm new to the library and am have been reviewing all of the issues posted here as well as the available documentation but still do not understand why I am having this issue. Can someone explain why applying the arm_nn_activations_direct_q15( ) using the tanh lookup table results in a zeroed out output for any number that is negative in the q15 format?

Thanks in advance for any help regarding this issue.

wwilliams3111, 2020-08-19:
So after some digging I believe there to be a bug in the `arm_nn_activations_direct_q15` function. Below is the code from lines 75-78 showing the process for indexing out the value from the lookup table. According to the documentation, it supports a range of input values from [-8,8); however, according to the code below, if you have a negative value, the unsigned saturation function will always saturate a negative value to 0, and thus index the first value in the lookup table which happens to be 0 for the TANH lookup table. My current workaround is to recast the value before entering the __USAT function to get the proper value from the lookup table. There may be a more elegant solution but so far this seems to work for my needs.

Original:
`        q15_t     in = *pIn++;
        q15_t     frac = (uint32_t) in & bit_mask;
        q15_t     value = lookup_table[__USAT((in >> shift_size), 8)];
        q15_t     value2 = lookup_table[__USAT((1 + (in >> shift_size)), 8)];
`
Workaround:
`        q15_t     in = *pIn++;
        q15_t     frac = (uint32_t) in & bit_mask;
        q15_t     value = lookup_table[__USAT((uint8_t)(in >> shift_size), 8)];
        q15_t     value2 = lookup_table[__USAT((uint8_t)(1 + (in >> shift_size)), 8)];
`

majianjia, 2020-08-26:
I think the table works well for negative values. 
Maybe your compiler setting or else cause this issue?

wwilliams3111, 2020-08-27:
You're right. I downloaded the developer version back in March and it looks like those changes weren't committed until April and I didn't noticed. It looks like they fixed several bugs in this version. Will update to the latest. Thanks

majianjia, 2020-08-28:
if you change `__USAT` to `__SSAT` might also work. It handles the out rangers compared to the current master branch.

wwilliams3111, 2020-08-28:
They use `__USAT` to keep the value positive for direct indexing of the lookup table. So if you don't recast the negative signed values to their unsigned counterpart which I ended up doing before I saw their fix, the `__USAT` function will just saturate all signed negative numbers to 0. 

They also address the condition when you index between the split positive and negative values of the lookup table and interpolate which produces the wrong result. Something I was also having an issue with in the older version.

closed 2020-08-31 by wwilliams3111