ARM-software/CMSIS_5 #986: Core: BPIALL register missing from some core register headers.

flit, 2020-08-18:
The `BPIALL` system register is missing from the `SCB_Type` struct in these headers:

- `core_armv8mml.h`
- `core_cm7.h`
- `core_cm33.h`

(It would be nice if we could separate the cache-management registers from `SCB_Type`, and not have duplicates of the MVFRx registers. But obviously that would not be backwards compatible.)

GuentherMartin, 2020-08-19:
core header files updated.
see commit https://github.com/ARM-software/CMSIS_5/commit/609db77b37ed85860f63cca9775a520d31fab5fd

flit, 2020-08-19:
Perfect, thanks!

JonatanAntoni, 2020-08-31:
Is this one already solved? Can we mark it "DONE" or even close it?

GuentherMartin, 2020-09-01:
BPIALL registers are added but no decision is made for:
"It would be nice if we could separate the cache-management registers from SCB_Type, and not have duplicates of the MVFRx registers. But obviously that would not be backwards compatible."

JonatanAntoni, 2020-09-01:
I think the answer is already included in the question: We should not go for incompatible changes. 
What could be done is duplicating all the cache registers into a separate struct. Would this be easier to understand and better aligned with the reference manual?

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni