ARM-software/CMSIS_5 #988: Improve the accuracy and range of arm_nn_activations_direct_q7?

majianjia, 2020-08-27:
https://github.com/ARM-software/CMSIS_5/blob/609db77b37ed85860f63cca9775a520d31fab5fd/CMSIS/NN/Source/ActivationFunctions/arm_nn_activations_q7.c#L60

https://github.com/ARM-software/CMSIS_5/blob/5eed9a31b456cc2cf485d9b8f8f73af6bebf51de/CMSIS/NN/Source/ActivationFunctions/arm_nn_activations_q7.c#L68

For both tanh and sigmoid. 
If I understand correctly, the table range it covers is from -8 to 8? (int bit=3), and the maximum int bit =3. 

The accuracy and range of tanh have causes some burden for me when I am trying to make an LSTM working. 
Most of the time the output of the cell state multiplication is in the range of Q2.5  but sometimes it requires Q4.3. The accuracy sometimes causes other issues when passing only Q0.7 data in it. 

Any plan or ideas to improve the range and accuracy of it? 
Thanks

wwilliams3111, 2020-08-28:
So I've been using the q15 version of this function. My understanding is the input of these functions should handle anywhere from Q0.n to Q3.n input type but you must specify the proper int width. Any values that are beyond Q3.n representation would saturate and approximate close to -1,0 or 1 depending on activation type and sign so they don't bother supporting beyond that input representation. What you can do is just apply a left shift to the data to represent Q3.n before passing through the activation function.

majianjia, 2020-08-28:
> So I've been using the q15 version of this function. My understanding is the input of these functions should handle anywhere from Q0.n to Q3.n input type but you must specify the proper int width. [...]

closed 2020-08-28 by majianjia