ARM-software/CMSIS_5 #994: Incorrect ARM Cortex M7 data cache manipulation code

dc42, 2020-09-07:
File CMSIS_5/CMSIS/Core/Include/cachel1_armv7.h includes the following line several times in functions that manipulate the data cache:

    SCB->CSSELR = 0U;                       /* select Level 1 data cache */

However, in documentation page https://developer.arm.com/documentation/ddi0489/b/system-control/register-descriptions/cache-size-selection-register it says that the instruction cache is selected if the LSB of CSSELR is zero, and this bit must be set to 1 to select the data cache. Is this a code bug or a documentation error?

JonatanAntoni, 2020-09-07:
Hi @dc42,

thanks for reporting this. This is apparently a bug in the code, good catch. Unfortunately, there seems not to be an easy way to test such pieces of code. Any ideas?

We will take a closer look and check how to fix it.

Cheers,
Jonatan

dc42, 2020-09-07:
To test it you first need a processor that has more sets or ways in the data cache than in the instruction cache, such as the ATSAME70 series processor that I am developing for (4-way data cache vs. 2-way instruction cache). Then you could write to a lot of cached memory; then call the function to disable the data cache. Then check whether the data has all been written back. Perhaps writing a block of memory equal to the data cache size would be sufficient to mark all cache lines dirty?

JonatanAntoni, 2020-09-07:
Hi @dc42,

that sounds easier as it might be. May I point you to [CMSIS/CoreValidation/Source/CV_CML1Cache.c](https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/CoreValidation/Source/CV_CML1Cache.c) which already contains some very basic test cases for cache functions on Cortex-M.

Ideally, we'd be able to add a unit test case for all the cache functions. Execution environment should be the [Fast Model](https://developer.arm.com/tools-and-software/simulation-models/fast-models) for Cortex-M7 (or M33 which has the same cache functionality).

Cheers,
Jonatan

fubo73, 2020-11-02:
> File CMSIS_5/CMSIS/Core/Include/cachel1_armv7.h includes the following line several times in functions that manipulate the data cache:
>
> ```
> SCB->CSSELR = 0U;                       /* select Level 1 [...]

The document you posted is old revision. Latest revision reports the correct behaviour of CSSELR register:
https://developer.arm.com/documentation/ddi0489/d/system-control/register-descriptions/cache-size-selection-register
So:
- 0 is for data cache
- 1 is for instruction cache

dc42, 2020-11-03:
Thanks for the clarification. However, this is a potential trap to developers (and it wasted several hours of my time) because a Google search for e.g "arm cortex m7 cache size selection register" returns the B version of the document. ARM should either remove the old version, or add a note to it that this part is incorrect.

fubo73, 2020-11-03:
The real problem is that you must know which is the revision of the HW. At each revision there is an associated TRM:
- r0p2 -> B
- r1p0 -> C
- r1p1 -> D
- r1p2 -> F
So, the code should identify the revision from CPUID before using the proper code.
In my case, for example, I have a r1p1 and the code in CMSIS is correct as is.

JonatanAntoni, 2021-02-08:
Hi @fubo73,

This relation between hardware revisions and document issues is not correct. Even if there are updates to the TRM aligned with the release of a new hardware revision, the updates apply to older hardware revisions as well.

According to the change log of the Cortex-M7 TRM the bit table given in issue B was simply wrong and was fixed in issue C:
>  CSSELR bit assignments table updated | Table 3.8 | All revisions

The CSSLER is architecturally defined and the [Armv7-M ARM](https://developer.arm.com/documentation/ddi0403/ed/) confirms that bit value 0 decodes data cache.

So, I think the implementation is CMSIS is correct for all hardware revisions.

Cheers,
Jonatan

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni