ARM-software/CMSIS_5 #995: Cortex M7 cache invalidation code doesn't fo last way or last set

dc42, 2020-09-07:
File CMSIS_5/blob/develop/CMSIS/Core/Include/cachel1_armv7.h contains several instances of this pattern, used to invalidate and/or flush the data cache:

    ccsidr = SCB->CCSIDR;
                                            /* invalidate D-Cache */
    sets = (uint32_t)(CCSIDR_SETS(ccsidr));
    do {
      ways = (uint32_t)(CCSIDR_WAYS(ccsidr));
      do {
        SCB->DCISW = (((sets << SCB_DCISW_SET_Pos) & SCB_DCISW_SET_Msk) |
                      ((ways << SCB_DCISW_WAY_Pos) & SCB_DCISW_WAY_Msk)  );
        #if defined ( __CC_ARM )
          __schedule_barrier();
        #endif
      } while (ways-- != 0U);
    } while(sets-- != 0U);

The loops terminate just after decrementing sets and ways to zero. So set 0 will never be processed, neither will way 0.

Suggestion: use this code instead:

    sets = (uint32_t)(CCSIDR_SETS(ccsidr)) + 1;
    do {
      --sets;
      ways = (uint32_t)(CCSIDR_WAYS(ccsidr)) + 1;
      do {
        --ways;
        SCB->DCISW = (((sets << SCB_DCISW_SET_Pos) & SCB_DCISW_SET_Msk) |
                      ((ways << SCB_DCISW_WAY_Pos) & SCB_DCISW_WAY_Msk)  );
        #if defined ( __CC_ARM )
          __schedule_barrier();
        #endif
      } while (ways != 0U);
    } while(sets != 0U);

JonatanAntoni, 2020-09-07:
Hi dc42,

thanks for reporting this back.
Did you recognize pre vs. post decrement in `while (ways-- != 0U)`? Actually, the loop should terminate if `ways` was zero before it got decremented (post decrement). I agree this can easily be misinterpreted.

The code has been written and manually optimized to result in efficient instructions on most compilers. Separating the decrement and adding `+1U` might be easier to read but could cause less efficient code (depending on the Compiler and used optimization level).

I'd prefer not to do such changes, easily. Please carefully check the original code one again if it really shows misbehavior in your environment. If the code actually behaves differently (i.e. not handling the post-decrement as I do expect) we'd need to check your environment (e.g. compiler and version). Please let me know.

Thanks again,
Jonatan

dc42, 2020-09-07:
I'm sorry, as you pointed out I made a mistake. Closing this.

closed 2020-09-07 by dc42