ARM-software/CMSIS_5 #998: A little attempt about accelerating the NN calculation

KevinLikesDringCoffe, 2020-09-16:
When I am implementing a fully-connected neural network on the EFM32LG(Cortex - M3 core), I found the matrix calculation can be accelerated by very simple modification. The key principle is to reduce the memory access. The original code is as below(only M0 and M3 branch).

`#else
    int       i, j;
    /* Run the following code as reference implementation for Cortex-M0 and Cortex-M3 */
    for (i = 0; i < num_of_rows; i++)
    {
        int       ip_out = ((q31_t)(bias[i]) << bias_shift) + NN_ROUND(out_shift);
        for (j = 0; j < dim_vec; j++)
        {
            ip_out += pV[j] * pM[i * dim_vec + j];
        }
        pOut[i] = (q15_t) __SSAT((ip_out >> out_shift), 16);
    }

#endif                          /* ARM_MATH_DSP */

    /* Return to application */
    return (ARM_MATH_SUCCESS);

}`
I tested it using a 30*28 layer and measured the excution time. The original code cost 0.794ms.
Then I modified the code. 
`int       i, j;
    int       rem;
    rem = num_of_rows % 2;
    /* Run the following code as reference implementation for Cortex-M0 and Cortex-M3 */
    for (i = 0; i < num_of_rows; i += 4)
    {
    	int       ip_out[2];
    	if(i + 2 > num_of_rows){
    		if(rem == 1){
    			ip_out[0] = ((q31_t)(bias[i]) << bias_shift) + NN_ROUND(out_shift);
    			for (j = 0; j < dim_vec; j++)
    			{
    				ip_out[0] += pV[j] * pM[i * dim_vec + j];
    			}
    			pOut[i] = (q15_t) __SSAT((ip_out[0] >> out_shift), 16);
    		}
    	}
    	else{
    		ip_out[0] = ((q31_t)(bias[i]) << bias_shift) + NN_ROUND(out_shift);
    		ip_out[1] = ((q31_t)(bias[i+1]) << bias_shift) + NN_ROUND(out_shift);
    		for (j = 0; j < dim_vec; j++)
    		{
    		    ip_out[0] += pV[j] * pM[i * dim_vec + j];
    		    ip_out[1] += pV[j] * pM[(i+1) * dim_vec + j];
    		}
    		pOut[i] = (q15_t) __SSAT((ip_out[0] >> out_shift), 16);
    		pOut[i+1] = (q15_t) __SSAT((ip_out[1] >> out_shift), 16);
    	}
    }
` 
by calculating two elements per iteration, we can reduce the memory access time, which is the main bottleneck in NN calculation on CPU. The excution time greatly dropped to 0.373ms

Then I wanted to futher improve the speed. I modified the code to calculating four elements per iteration. But I found it was slower than the previous attempt which calculated two elements per iteration. I thought it might due to the limited registers in the M3 architecture and there were additional codes.

Here is my experimental code
`/*
 * Copyright (C) 2010-2018 Arm Limited or its affiliates. All rights reserved.
 *
 * SPDX-License-Identifier: Apache-2.0
 *
 * Licensed under the Apache License, Version 2.0 (the License); you may
 * not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 * www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
[... 261 lines cut ...]
    		}
    	}
    	else{
    		ip_out[0] = ((q31_t)(bias[i]) << bias_shift) + NN_ROUND(out_shift);
    		ip_out[1] = ((q31_t)(bias[i+1]) << bias_shift) + NN_ROUND(out_shift);
    		for (j = 0; j < dim_vec; j++)
    		{
    		    ip_out[0] += pV[j] * pM[i * dim_vec + j];
    		    ip_out[1] += pV[j] * pM[(i+1) * dim_vec + j];
    		}
    		pOut[i] = (q15_t) __SSAT((ip_out[0] >> out_shift), 16);
    		pOut[i+1] = (q15_t) __SSAT((ip_out[1] >> out_shift), 16);
    	}
    }

#endif

    return (ARM_MATH_SUCCESS);

}

/**
 * @} end of FC group
 */
`

majianjia, 2020-09-17:
This part of code is for reference. 
If I understand correctly, you are doing an unroll-of-loops, which compiler might also do if you increase your compiler's optimation level. 
Did you try different optimation level?

felix-johnny, 2020-09-23:
@KevinLikesDringCoffe  Thanks for the ticket. you are right about the reduced memory accesses as that is usually what is done in most optimizations combined with a constraint on the number of General Purpose registers.
Shouldn't the i += 4 be i += 2 instead? 

Not sure if the compiler would loop unroll the outer loop here with a different optimization level.

felix-johnny, 2020-09-28:
You are welcome to make a PR on this. I would like the structure to be as two distinct blocks though. Something like

     int32_t row_loop_cnt = num_of_rows / 2;
    for (int i_row_loop_cnt = 0; i_row_loop_cnt < row_loop_cnt; i_row_loop_cnt++)
   {
    Handle even number of rows.
   }
   
   row_loop_cnt = num_of_rows & 0x1;
   if (row_loop_cnt)
 {
  // Handle left over row
 }

 With that, I'll close this ticket. Thanks again for the input.

closed 2020-09-28 by felix-johnny