ARM-software/CMSIS_5 #1001: CMSIS-DSP: Redundant expressions about arm_bayes_example

dangorogoro, 2020-09-24:
According to the function of **arm_gaussian_naive_bayes_predict_f32**, it puts each probability to the buffer and returns the index of the highest probability value.

But in the example code of **arm_bayes_example_f32.c**, it calculates the max value in the result buffer even though **arm_gaussian_naive_bayes_predict_f32** returns it.
```c
  arm_gaussian_naive_bayes_predict_f32(&S, in, result);

  arm_max_f32(result, NB_OF_CLASSES, &maxProba, &index);
```

I think this is a bit redundant. So, unless there is a clear reason to write it this way, wouldn't it be less confusing to write the following?
```c
  index = arm_gaussian_naive_bayes_predict_f32(&S, in, result);

  maxProba = result[index];
```
I've written a pull request, so please take a look at that as well.

Thank you.

christophe0606, 2020-09-24:
Hi @dangorogoro , 

You're totally right !

It is redundant. In the example I wanted to extract the probability in addition to the index.
But it does not make sense to re-compute the max for this.

christophe0606, 2020-09-24:
I'll merge your pull request a bit later.

PR #1002 (2020-09-25): CMSIS-DSP: Modified the arm_bayes_example
Closes #1001
**arm_gaussian_naive_bayes_predict_f32** already returned index, modified the example to remove the redundant expressions.
closed 2020-09-25 by christophe0606