ARM-software/CMSIS_5 #1003: What is "compilerSpecificCompileOptions" in configCore.cmake

a76827, 2020-09-25:
Hi, 
  I am trying to build CMSIS-DSP and NN library using CMake on Ubuntu 16.04 LTS and CMake version 3.17
When I type 

**cmake -DCMAKE_PREFIX_PATH="/home/toolchain/arm-none-eabi/" \
      -DCMAKE_TOOLCHAIN_FILE="../../../DSP/gcc.cmake" \
      -DARM_CPU="cortex-m3" \
      -DROOT="../../../.." \
      -G "Unix Makefiles" ..**

the error report as :
**CMake Error at /home/Software_framework/CMSIS_5/CMSIS/DSP/configCore.cmake:295 (compilerSpecificCompileOptions):
  Unknown CMake command "compilerSpecificCompileOptions".
Call Stack (most recent call first):
  /home/Software_framework/CMSIS_5/CMSIS/DSP/configLib.cmake:27 (configcore)
  BasicMathFunctions/CMakeLists.txt:34 (configLib)**

I searched whole CMSIS and googled "compilerSpecificCompileOptions" but found nothing.
Does anyone can help me to fix this error?

Thanks

christophe0606, 2020-09-25:
Note that cmake is not the officially supported way of building CMSIS. The MDK or ArmDS should be used.

I am using it for DSP. For NN, I have not been maintaining it so it may no more work.

The compilerSpecificCompileOptions is defined in DSP/Toolchain.
There is one file per supported compiler. 

And in gcc.cmake for instance you have a definition of compilerSpecificCompileOptions.

The goal of those function is to translate some cmake options into command line options for a specific compiler.

For instance FASTMATHCOMPUTATIONS to -ffast-math

So, your error means the file was not loaded and the function is not defined.

If you want to build only CMSIS-DSP, or an app using CMSIS-DSP with cmake, you'll need to write a cmake as explained at the end of the DSP/README.md : "Building only the CMSIS-DSP library"

In your command I don't know where is the ".." pointing at the end of the command ? It would help to understand the issue.

a76827, 2020-09-28:
Hi @christophe0606 ,
    Thanks for your answer, the CMake command I use is in NN folder, so ".." points to CMSIS/NN/Source , I create a build folder and execute above command.
   And the README.md points CMAKE_TOOLCHAIN_FILE to /CMSIS/DSP/gcc.cmake instead of /CMSIS/DPS/Toolchain/GCC.cmake, How and when do I include these 2 files?

Thanks again

christophe0606, 2020-09-29:
Hi @a76827 ,

It won't work. You need to create a CMakeLists.txt as explained in the README.
The README is giving an example for DSP but it is the same for NN. You just need to load NN folder instead.

As I said : NN cmake are not maintained. So it may not work with a few additional fixes. But if you can build DSP it means at least your cmake config is right.

christophe0606, 2020-09-29:
Create a project folder. Inside create a CMakeLists.txt as explained below.
Then, create a build folder inside this project folder.
From the build folder, type the usual cmake command.

`cmake_minimum_required (VERSION 3.6)

### Define the project
project (testcmsisdsp VERSION 0.1)

### Define the path to CMSIS-DSP (ROOT is defined on command line when using cmake)
set(DSP ${ROOT}/CMSIS/DSP)

### Add DSP folder to module path
list(APPEND CMAKE_MODULE_PATH ${DSP})

### Load CMSIS-DSP definitions. Libraries will be built in bin_dsp
add_subdirectory(${DSP}/Source bin_dsp)
`

a76827, 2020-10-06:
Hi @christophe0606 ,
   Thanks for your advice, I can build both CMSIS-DSP and CMSIS-NN library now.

closed 2020-10-06 by a76827