ARM-software/CMSIS_5 #1006: GIC: IRQ_ID0 in workaround for errata 733075 possibly ineffective

alibabashack, 2020-09-29:
The function IRQ_GetActiveIRQ()
https://github.com/ARM-software/CMSIS_5/blob/e47704d3004373b3fe867906d0baa2a70d0995a1/CMSIS/Core_A/Source/irq_ctrl_gic.c#L239

includes a workaround for
> GIC 390 errata 733075 (GIC-390_Errata_Notice_v6.pdf, 09-Jul-2014)

Apparently, the workaround tries to remember if an IRQ with id 0 is active. It is unclear to me why, because I cannot find any document containing details on GIC 390 errata 733075. However, to my understanding, the introduced variable IRQ_ID0 does not change the control flow in any way.

The line
https://github.com/ARM-software/CMSIS_5/blob/e47704d3004373b3fe867906d0baa2a70d0995a1/CMSIS/Core_A/Source/irq_ctrl_gic.c#L265
only guards if IRQ_ID0 is set to 1 and no other code depends on the value of IRQ_ID0. So IRQ_ID0 is a self fulfilling prophecy.

1. Please provide a source for the referenced errata (GIC 390 errata 733075).
2. Please explain what IRQ_ID0 is supposed to do.

VladimirUmek, 2020-10-01:
This requires a bit of investigation but the purpose of this handling was basically to filter IRQ ID 0 as spurious interrupt or as valid interrupt request.
I agree that current implementation does not do anything and needs to be fixed. But the code and reasoning must be first reviewed in detail.

alibabashack, 2020-10-02:
Thanks for reviewing. Any thoughts where to obtain the referenced errata description (GIC 390 errata 733075)?