ARM-software/CMSIS_5 #1008: CMSIS-NN impossible constraint in 'asm'

janjongboom, 2020-09-30:
I'm using CMSIS5 rev 01f5b32badf7b78c85a24a7149b56400fa6a2999 with both CMSIS-Core and CMSIS-NN on a STM32 Cortex-M4F. We have a working project that builds fine on many targets, including on Mbed OS on the same platform (GCC ARM 9).

However building with STM32CubeIDE (both GCC7 and GCC9) fails with:

```
../Drivers/CMSIS/Include/cmsis_gcc.h:41:50: error: impossible constraint in 'asm'
   41 |   #define __ASM                                  __asm
      |                                                  ^~~~~
../gestures/edge-impulse-sdk/CMSIS/NN/Source/NNSupportFunctions/arm_nn_mat_mult_nt_t_s8.c:38:1: note: in expansion of macro '__ASM'
   38 | __ASM ("sxtb16 %0, %1, ROR %2" : "=r" (result) : "r" (op1), "i" (rotate) );
```

Is there a compiler flag that I'm missing?

felix-johnny, 2020-10-01:
@janjongboom Please have a look at https://github.com/ARM-software/CMSIS_5/issues/996 for the solution to this issue. The PR to fix this issue is not made yet and is in the making.

JonatanAntoni, 2020-10-01:
Hi @janjongboom,

do you run this on Mac like in #996? It looks like GCC behaves differently on Mac.

Cheers,
Jonatan

janjongboom, 2020-10-01:
@felix-johnny Thanks! 

@JonatanAntoni Yes, this is using `arm-none-eabi-gcc (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1 20200408 (release)` on MacOS.

janjongboom, 2020-10-01:
Indeed same solution works on our end as in #996.

felix-johnny, 2020-10-13:
Fixed by https://github.com/ARM-software/CMSIS_5/commit/3551ea849669765339f7a5b0400472eb23f6731a .

closed 2020-10-13 by felix-johnny