ARM-software/CMSIS_5 #1011: Consider removing the register keyword from core_cm7.h

syntroniks, 2020-10-04:
Compiling C++ programs which leverage CMSIS will bring in core-specific headers.

In my case, when C++17 standard support is enabled, GCC emits the following warnings:

```
[build] ../extern/sdk/CMSIS/Include/core_cm7.h: In function 'void SCB_DisableDCache()':
[build] ../extern/sdk/CMSIS/Include/core_cm7.h:2368:23: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
[build]  2368 |     register uint32_t ccsidr;
[build]       |                       ^~~~~~
[build] ../extern/sdk/CMSIS/Include/core_cm7.h:2369:23: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
[build]  2369 |     register uint32_t sets;
[build]       |                       ^~~~
[build] ../extern/sdk/CMSIS/Include/core_cm7.h:2370:23: warning: ISO C++17 does not allow 'register' storage class specifier [-Wregister]
[build]  2370 |     register uint32_t ways;
[build]       |                       ^~~~
```

Given that the _C++_ compiler is allowed to ignore `register`,
• and that it is presumably important that these variables are supported with registers (for cache disable),
• and that CMSIS aims to write C as opposed to assembly,

what are we to do?

JonatanAntoni, 2021-06-08:
@syntroniks,

The code under question has been moved to `cachel1_armv7.h` and should be fine for C and C++ use.
May I ask you to double check and close this issue, please?

Thanks,
Jonatan

syntroniks, 2021-06-08:
Oh certainly, this looks wonderful. Thank you for working on this 👍

closed 2021-06-08 by syntroniks