ARM-software/CMSIS_5 #1012: arm_avgpool_s8 function seems to have offset of 1/2 for positive or -1/2 for negative values, why ?

tkafafi, 2020-10-08:
In CMSIS/NN/Source/PoolingFunctions/arm_avgpool_s8.c line 334 below:      
 sum = sum > 0 ? (sum + count / 2) / count : (sum - count / 2) / count;

Adding or subtracting count/2, basically adds 1/2 if average is positive or subtracts 1/2 if average if negative. 
What is the purpose of this ?

majianjia, 2020-10-09:
It is a simple trick for converting the `floor()` into `round()`.

example:
sum = 3, count =4
without `+count/2` the line is equal to `average = 3/4 =0` which is a `floor(3.0 / 4.0)`
with `+count/2` then it is `average = (3+2) / 4 = 1` which is a `round(3.0 / 4.0)`

tkafafi, 2020-10-09:
Thanks for the explanation Jianjia.

I believe there is an issue with negative numbers exactly at half point. The current implementation with round the negative numbers at half point downwards, towards the more negative value. Generally, I expect the number at half point to be rounded upwards, ie towards the less negative number .
eg. sum = -10, count = 4.
-10/4 = -2.5 which should be rounded to -2.
the current formula of (sum-count/2)/count will result in (-10 - 2)/4 = -3

This can be fixed by using (sum - (count -1)/2)/count.
This will result in (-10 -3/2)/4 = -11/4 = -2

Thanks

majianjia, 2020-10-10:
I think `round()` is round down for negative num. And the code is implementing round()

felix-johnny, 2020-10-15:
@majianjia Thanks for the clarifications :+1: and @tkafafi for the question!   It is 'round with midpoint away from zero'. Here is another instance where it is used https://github.com/ARM-software/CMSIS_5/blob/318e309ade12fd46d9d3093727c66f3fe550946d/CMSIS/NN/Include/arm_nnsupportfunctions.h#L740 
 
 Let me know if this clarifies.

closed 2021-01-07 by felix-johnny