ARM-software/CMSIS_5 #1015: CMSIS-DSP & NN: Compiling Pure C version to simulate on PC

justiceamoh, 2020-10-12:
Hi, 

I am trying to compile pure c versions of CMSIS DSP (and CMSIS-NN) to simulate and test models on my laptop (macOS) before deployment to hardware. I saw on #490 that this is possible, but I'm struggling setting up the build environment for successful compilation. I followed the README in the DSP directory and from #1003, and I created this minimal CMakeList for configuring the build:

```cmake
cmake_minimum_required (VERSION 3.14)

# Define compiler as homebrew gcc, instead of clang
set(CMAKE_C_COMPILER "gcc-10")
set(CMAKE_CXX_COMPILER "g++-10")

# Define the path to CMSIS-DSP
set(ROOT /path/to/repo/CMSIS_5)
set(DSP ${ROOT}/CMSIS/DSP)
set(CMAKE_TOOLCHAIN_FILE ${DSP}/Toolchain/GCC.cmake)
set(ARM_CPU "cortex-m0")

# Define the project
project (testcmsisdsp)

# Add DSP folder to module path
list(APPEND CMAKE_MODULE_PATH ${DSP})

# Load CMSIS-DSP definitions.
add_subdirectory(${DSP}/Source bin)
```
I run into some compile errors: unrecognized command-line option '-mlittle-endian' and unrecognized command-line option '-mthumb', which seems to indicate the sources are still expecting an arm compiler, even though the target is set as `cortex-m0`.

For CMSIS-NN, #484 asserts that it's pure C, but there's a heavy reliance on `arm_math.h`, which has now been refactored and is difficult to remove. I see there are the reference implementations but they also seem to depend on `arm_math.h`. 

Can you please advise on the best approach to go about this?

christophe0606, 2020-10-13:
@justiceamoh If you want to compile on a host (like Mac), you must not use the ${DSP}/Toolchain/... files since they are for cross compilation only.

Following CMakeLists.txt should work (I have not tested but with a bit of tuning it should work):

```
cmake_minimum_required (VERSION 3.6)
project (testcmsisdsp VERSION 0.1)

add_executable(testcmsisdsp main.c)

set(DSP ${ROOT}/CMSIS/DSP)
list(APPEND CMAKE_MODULE_PATH ${DSP})
add_subdirectory("${DSP}/Source" bin_dsp)

target_compile_definitions(CMSISDSP __GNUC_PYTHON__)

target_link_libraries(testcmsisdsp PRIVATE CMSISDSP)

```

The __GNUC_PYTHON__ is required because the cortex-m0 code is not pure C. It still contains a few intrinsics like SSAT. For the Python wrapper (allowing to run CMSIS-DSP with numpy), we had introduced the possibility to build on host by redefining in C those intrinsics.

Now for NN, you just need to have the add_subdirectory for NN and target_link_libraries for NN.

Those cmake were introduced for the DSP test framework. NN is not maintained as often so there may be a few additional problems.

I am not sure that the GNUC_PYTHON will be applied to the subprojects of CMSISDSP by doing like that.
So perhaps you may want to use:

```
SET(CMAKE_C_FLAGS_DEBUG "-D__GNUC_PYTHON__"  CACHE INTERNAL "")

```

(Or _RELEASE if you build in Release mode with cmake).

I'll add a cmake option to make it easier to build on host and hide this GNUC_PYTHON.

justiceamoh, 2020-10-13:
Thanks so much for getting back to me @christophe0606. I have been able to successfully compile CMSIS DSP using the `__GNUC_PYTHON__` flag. And you were right, adding that in `target_compile_definitions` does not propagate to the subprojects. So I needed to prefill the cache before the `project` command.  I also found it was necessary to a toolchain file so I used the GCC one like so: `set(CMAKE_TOOLCHAIN_FILE ${DSP}/Toolchain/GCC.cmake)`.

I followed your suggestions for building CMSIS-NN and sure enough, I hit some of those problems you alluded to. The error I had was in `arm_nnsupportfunctions.h`:

```
CMSIS_5/CMSIS/NN/Include/arm_nnsupportfunctions.h:752:58: error: expected ')'
__STATIC_FORCEINLINE void arm_memcpy_q7(q7_t *__RESTRICT dst,
```
I think this is because `__RESTRICT` is not defined even with the `__GNUC_Python__` compile definition. I am able to successfully compile if I remove the `__RESTRICT`  from `arm_memcpy` definition in `arm_nnsupportfunctions.h`. 

`__RESTRICT` is defined in `cmsis_compiler.h`. For compilers that don't support it, it's just ignored by: 
```c
  #ifndef   __RESTRICT
    #warning No compiler specific solution for __RESTRICT. __RESTRICT is ignored.
    #define __RESTRICT
  #endif
```
So perhaps you can include that in your definitions for `__GNUC_Python__`. Thank you once again for the support!

christophe0606, 2020-10-14:
Can you tell me what problem you had when you don't use 

`set(CMAKE_TOOLCHAIN_FILE ${DSP}/Toolchain/GCC.cmake`

Because this file is defining options which make sense only if you build for ARM. So if you build for host, it may cause problems.

Normally this file is not needed.

justiceamoh, 2020-10-14:
Sure. Without specifying the toolchain, I get the following error when I try to configure cmake:

```bash
CMake Error at CMSIS_5/CMSIS/DSP/configCore.cmake:213 (compilerSpecificCompileOptions):
  Unknown CMake command "compilerSpecificCompileOptions".
Call Stack (most recent call first):
  CMSIS_5/CMSIS/DSP/configLib.cmake:27 (configcore)
  CMSIS_5/CMSIS/DSP/Source/BasicMathFunctions/CMakeLists.txt:12 (configLib)

-- Configuring incomplete, errors occurred!
See also "/Users/Junior/repos/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeOutput.log".
See also "/Users/Junior/repos/CMSIS_5/CMSIS/DSP/BuildCMSISOnly/build/CMakeFiles/CMakeError.log".
```

I think it results from the include `CMakeLists.txt` in the `DSP/Source` folder, which calls `configLib` cmake function, which would typically require the toolchain.

christophe0606, 2020-10-14:
Thanks. I'll look at it to make it easier to build on host and why it works on my side (On windows using MSVC). 

I suggest to define an empty function in your CMakeLists.txt file:

```
function(compilerSpecificCompileOptions PROJECTNAME ROOT)
endfunction()
``` 

rather than including the Toolchain because the toolchain is for cross compilation.
This function is not useful for building on host.

justiceamoh, 2020-10-15:
Thanks for that suggestion. Using the empty function works and is much cleaner than including the toolchain file. I am on a macOS host so perhaps the MSVC flag deactivates certain features that enables the build to work on your side.

Thanks for all your support. I'm able to test both CMSIS-DSP and CMSIS-NN functions on my laptop now.

closed 2020-10-15 by justiceamoh

sigurdpl, 2021-03-16:
Hi, i am also building on MacOS. I followed the advice above but get this error while building : error: implicit declaration of function '__ROR' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
    *pDst++ = (__ROR(in, 8)) & 0xFF000000;
from CMSIS_5-5.7.0/CMSIS/DSP/Source/SupportFunctions/arm_q7_to_q31.c:120:

christophe0606, 2021-03-16:
@sigurdpl Can you open a new issue ?
In the new issue can you tell me the compilation flag you see when building arm_q7_to_q31.c.

EdXian, 2021-11-25:
Hi @justiceamoh,

I have built the CMSIS-DSP package successfully on my linux PC(x86). However,  I got some compiling  error when I include the source file from NN package. The system returns error which as same as yours 
```
CMSIS_5/CMSIS/NN/Include/arm_nnsupportfunctions.h:752:58: error: expected ')'
__STATIC_FORCEINLINE void arm_memcpy_q7(q7_t *__RESTRICT dst,
```
I removed the macro `__RESTRICT` but other error still exists.  Some error is caused by using assembly in function. And the assembly is only for ARM Cortex-M. 
Do you have any idea?

If you see this comment, please refer to the new issue https://github.com/ARM-software/CMSIS_5/issues/1373