ARM-software/CMSIS_5 #1018: FFT comparison between Matlab and CMSIS DSP Library

razackkpa, 2020-10-14:
We have done FFT on 4096 data samples at a rate of 20K and done FFT in an embedded device using CMSIS DSP Library (arm_cfft_radix4_init_f32.c) and used the same 4096 data and did FFT using Matlab. We are getting two different graphs (graphs attached)

Version V.1.4.1 

Dated - 17 Jan 2013

Please help clarify this. Appreciate your help!

[image]
[image]

christophe0606, 2020-10-14:
@razackkpa  Note that the arm_cfft_radix4_f32 is deprecated and should no more be used.
Instead use arm_cfft_f32

Would it be possible for you to share your input signal so that I can try to reproduce the problem ?

Also what are the values you pass to arm_cfft_radix4_init_f32

Thanks.

razackkpa, 2020-10-19:
Hi Christ,
Appreciate you help.
Please see the input signal

Time | Amps
-- | --
0 | 345
0.00005 | 3240
0.0001 | 2881
0.00015 | 2836
0.0002 | 2739
0.00025 | 2926
0.0003 | 2775
0.00035 | 3067
0.0004 | 2674
0.00045 | 2878
0.0005 | 2688
0.00055 | 2814
0.0006 | 2952
0.00065 | 2823
0.0007 | 2922
0.00075 | 2649
0.0008 | 2763
0.00085 | 2853
0.0009 | 2707
0.00095 | 2735
0.001 | 2781
0.00105 | 2910
0.0011 | 2851
0.00115 | 2849
0.0012 | 2907
0.00125 | 2772
0.0013 | 3002
0.00135 | 2984
0.0014 | 2992
0.00145 | 2950
0.0015 | 2782
0.00155 | 3035
0.0016 | 2949
0.00165 | 2953
0.0017 | 2969
0.00175 | 2912
0.0018 | 2920
0.00185 | 2923
0.0019 | 3070
0.00195 | 3024
0.002 | 3021
0.00205 | 3062
0.0021 | 3002
0.00215 | 3048
0.0022 | 2981
0.00225 | 3010
0.0023 | 2989
0.00235 | 3037
0.0024 | 2738
0.00245 | 3117
0.0025 | 3486
0.00255 | 2782
0.0026 | 3003
0.00265 | 3217
0.0027 | 3160
0.00275 | 3150
0.0028 | 3038
0.00285 | 2818
0.0029 | 3899
0.00295 | 2807
0.003 | 2950
0.00305 | 3091
0.0031 | 3085
0.00315 | 2847
[... 4007 lines cut ...]
0.20355 | 859
0.2036 | 1038
0.20365 | 940
0.2037 | 829
0.20375 | 891
0.2038 | 918
0.20385 | 800
0.2039 | 792
0.20395 | 990
0.204 | 827
0.20405 | 705
0.2041 | 787
0.20415 | 654
0.2042 | 562
0.20425 | 765
0.2043 | 556
0.20435 | 669
0.2044 | 546
0.20445 | 576
0.2045 | 516
0.20455 | 395
0.2046 | 611
0.20465 | 410
0.2047 | 487
0.20475 | 585

christophe0606, 2020-10-19:
Hi @razackkpa,

I have made 3 tests : Matlab, Mathematica and CMSIS-DSP.
I get the same result in the 3 cases.

**Matlab:**
(I have rescaled time between 0 and 1 for the plot)

[image]

**Mathematica:**

[image]

**CMSIS-DSP:**

[image]

The value in 0 is : 8188420 for the 3 tests.
The value in 4095 is : 1.3123e+05 + 3.8078e+04i for the 3 tests.

So I don't see any problem in the CMSIS-DSP.

Are you using a complex signal in input of CMSIS-DSP ? 
Like 345 0 3240 0 2881 0 ...

Are you enabling the bit reverse when using the CMSIS-DSP FFT ?

christophe0606, 2020-10-19:
I get the same results with the radix4 version.

razackkpa, 2020-10-21:
Please see the part of code we use for the same input given to you. 

FFT_SIZE = 4096
float32_t Input[FFT_SIZE];
float32_t Output[FFT_SIZE];

FFT_SIZE = 4096
float32_t Input[FFT_SIZE];
float32_t Output[FFT_SIZE];

arm_cfft_radix4_init_f32(&S, FFT_SIZE, 0, 1);
        arm_cfft_radix4_f32(&S, Input);
        arm_cmplx_mag_f32(Input, Output, FFT_SIZE);
        arm_max_f32(Output, FFT_SIZE, &maxValue, &maxIndex);

christophe0606, 2020-10-21:
@razackkpa cfft is using a complex signal in entry.
So, if your FFT is 4096 real samples, then you need an array of 8192 complex samples at the input (with imaginary part set to 0).

Your arrays Input and Output are too small. They both need to be 8192.
FFT_SIZE is 4096.

The output of arm_cmplx_mag_f32 cannot be the 8192 Output array. Because the output of this function is real. So in this case you'll need a 4096 array for the magnitudes.

razackkpa, 2020-10-21:
input is a float array, so in order to get FFT is 4096 the input size should be 8192 by appending 4096. This 0 should be alternate or need to append at the end?
Please clarify

christophe0606, 2020-10-21:
@razackkpa Alternate : a 0 b 0 c 0 ...

christophe0606, 2020-10-22:
You may want to look at the example DSP/Examples/ARM/arm_bin_fft_example.
TEST_LENGTH_SAMPLES is twice the fftSize in the example.

I am closing this issue now since there is no bug.

closed 2020-10-22 by christophe0606

razackkpa, 2020-11-16:
Hi Christ-
Working fine!
Appreciate your help.