ARM-software/CMSIS_5 #1021: Include directories not added in GCC build using CMSIS build

mentaal, 2020-10-19:
I can successfully build my project using CMSIS build with the AC6 toolchain.
When I try and use GCC, the makefile is missing some include paths. 

The `.cprj` file has the required include directories in the `cflags` element.

When using AC6, I can see these include directories inside `CC_FLAGS_GLOBAL` in the target makefile. When using GCC, this variable isn't present and the `INC_PATHS` doesn't have the directories in there either. (Well it has some such as for CMSIS and the Keil packs, just not my application include directories).

brondani, 2020-10-21:
Hi Gregory,

As you have observed, the `cflags` content at target level will be set as `CC_FLAGS_GLOBAL` in the makefile and they are specific for a chosen compiler/toolchain.
Can you please make sure the [cflags element](https://arm-software.github.io/CMSIS_5/Build/html/element_target.html#element_target_cflags) has the correct `compiler` attribute, in your case `compiler="GCC"` instead of `compiler="AC6"`?
If it does not solve the issue, can you please provide a project example?

Thanks
Daniel

mentaal, 2020-10-21:
Hi Daniel, 

Thanks for the prompt response. 
Oh right I understand. Your intuition is correct and I had not noticed this within the `.cprj` file. 

I wonder, am I using the tool differently to how it was intended to be used?
What I did was manually add the GCC entry below into the compilers section of the file as in below:

```
  <compilers>
    <compiler name="AC6" version="6.0.0:6.99.99"/>
    <compiler name="GCC" version="9.2.1:9.2.1"/>
  </compilers>
```
I also hadn't noticed the `compiler="AC6"` attribute of the `cflags` element. 
Manually changing this compiler option to be GCC does indeed fix this issue in that CC_FLAGS_GLOBAL is now present but of course the flags are all intended for AC6 as I'm probably clumsily hacking the .cprj file.

brondani, 2020-10-22:
Your `compilers` entries are saying the project can be compiled with AC6 (any version between 6.0.0 and 6.99.99) or GCC (9.2.1).
Note that you can have multiple cflags elements, one for each supported toolchain. For example:
```
<target ...
  <cflags add="..." compiler="AC6"/>
  <cflags add="..." compiler="GCC"/>
</target>
```

mentaal, 2020-10-22:
Great stuff. Thanks Daniel.

closed 2020-10-22 by mentaal