ARM-software/CMSIS_5 #1024: X use VFP register arguments, Y does not query

mentaal, 2020-10-22:
Hi All, 
I am seeing this error when compiling with GCC using cbuild but I don't get this error when compiling using AC6.

From what I can understand from various sources such as:

1. https://embeddedartistry.com/blog/2017/10/11/demystifying-arm-floating-point-compiler-options/
2. https://stackoverflow.com/questions/9753749/arm-compilation-error-vfp-registers-used-by-executable-not-object-file

This would seem to be that I am building my application with an incompatible FPU ABI as compared to a dependency library I'm trying to link with.

Ok, that's understood, but then, how is this working with AC6? 

In both cases, GCC and AC6, the CPU, FPU is set thusly in the generated makefile:

```
CPU:=Cortex-M7
FPU:=DP_FPU
```

I'm attempting to use GCC 9.2.1.

In `cbuild\etc\*.mak` files I can see:

GCC.9.2.1.mak:

```
else ifeq ($(CPU),Cortex-M7)
  ifeq ($(FPU),DP_FPU)
    GNUASM_CPU:=-mcpu=cortex-m7 -mfpu=fpv5-d16 -mfloat-abi=hard
  else ifeq ($(FPU),SP_FPU)
    GNUASM_CPU:=-mcpu=cortex-m7 -mfpu=fpv5-sp-d16 -mfloat-abi=hard
  else
    GNUASM_CPU:=-mcpu=cortex-m7
  endif

```

ARMCC.6.14.0.mak:

```
else ifeq ($(CPU),Cortex-M7)
  ifeq ($(FPU),DP_FPU)
    ARMCLANG_CPU:=-mcpu=$(CPU) -mfpu=fpv5-d16 -mfloat-abi=hard
  else ifeq ($(FPU),SP_FPU)
    ARMCLANG_CPU:=-mcpu=$(CPU) -mfpu=fpv5-sp-d16 -mfloat-abi=hard
  else
    ARMCLANG_CPU:=-mcpu=$(CPU) -mfpu=none
  endif

```

Which looks basically the same.

Thanks.

brondani, 2020-10-27:
Hi Gregory,

When linking hard-float objects for soft-float targets, armlink will probably throw this [kind of message](https://developer.arm.com/documentation/100074/0615/Linker-Errors-and-Warnings/List-of-the-armlink-error-and-warning-messages):
`L6366E: <object> attributes<attr> are not compatible with the provided cpu and fpu attributes<cli> <diff>.`

On the other way around, linking soft-float objects for hard-float targets as in your example, armlink seems to be more permissive than GCC and allows the linking step, however the generated executable will work correctly only if the functions at the objects' public interfaces are compatible. Indeed hard-float objects may pass arguments in FPU registers while soft-float objects use CPU registers.

The [Application Note 487](https://developer.arm.com/documentation/dai0487/a/) gives more info about binary interoperability with focus on different toolchains.  

Best regards
Daniel

mentaal, 2020-10-27:
Thanks for the reply Daniel and the link. 

Ok understood. We run the risk of the building an incompatible binary in the case of armlink. 
Would you happen to know, is it possible to link an application using GCC in this case? 
I did try manually changing the `mfloat-abi` argument within `GCC.9.2.1.mak` to be `softfp` instead of `hard` as was the default but this just resulted in link errors because it's trying to link against a library using the hard ABI:

```
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe:
error: C:/Users/gkuhn/AppData/Local/Arm/Packs/ARM/CMSIS/5.7.0/CMSIS/RTOS2/RTX/Library/GCC/libRTX_CM4F.a(rtx_system.o) uses VFP register arguments, <my_target_application.elf> does not
```

I guess it would be left to me to compile RTX5 from source as opposed to using the provided library. If that's the solution, then that's fine and this ticket can be closed. 

PS, I'm aware that this question is probably a bit out of scope of this project at this point so I'm very thankful for the help you've provided me here.

brondani, 2020-10-27:
You are right, a possible solution for your case is to recompile the source code with softfp, which means it will use hardware floating-point instructions but software floating-point linkage.

An alternative is to set the target without FPU in the cprj file:
`<target Dfpu="NO_FPU" Dname="ARMCM7" ...>`
in such case the project is built with plain software library functions for floating-point operations and a pre-compiled software floating-point RTX library is selected: `libRTX_CM3.a`

If you choose to customize the toolchain config file `GCC.9.2.1.mak`, note that you can put the custom file in the project directory and it will have precedence in respect to the file in the `$CMSIS_COMPILER_ROOT` directory.

I'm glad this info has been helpful to you, feel free to open issues whenever you think it's worth.

closed 2020-10-27 by brondani

mentaal, 2020-10-27:
Yup this is very helpful thank you
I also wasn't aware of the "NO_FPU" option which I will definitely investigate. 
This option is probably what I'll go for as it sounds like the simplest. I understand I'll be sacrificing performance but I don't think it it will matter in my case as no floating point arithmetic is being done in main library I'm concerned with or the application code around it.

Cheers