ARM-software/CMSIS_5 #1025: Possible issue with CMSIS IIR functions and DS memory?

raysimar, 2020-10-22:
Hi there!

I have been having an unusual problem with the attached file.  I have pruned it down to the critical bits to reproduce the problem.  It's a test bench, similar to the FIR example as shown in CMSIS Examples.

Mine is testing the functions:
- arm_biquad_cas_df1_32x64_init_q31(&S, NUM_STAGES, IIRCoeffs32, IIRState64, postShift);
- arm_biquad_cas_df1_32x64_q31 (&S, testInputi32 + (i * blockSize), testOutputi32 + (i * blockSize), blockSize);

What is happening is that, if I use this macro setting of 40, things appear to work fine.
#define TEST_LENGTH_SAMPLES  40 // Pruning this to 320
 
If I change it to 48, it freezes.
 
This is a pruned down version of the original program.  I am suspicious that I am somehow corrupting the memory/stack.
 
Could you take a look and let me know what you think?  I am happy to have a chat about this too!
 
Thanks!
 
All the best,
Ray Simar
Professor in the Practice
Rice University

[main.txt](https://github.com/ARM-software/CMSIS_5/files/5422533/main.txt)

christophe0606, 2020-10-23:
@raysimar It looks like the vectorized implementation is not working when blockSize <= 3.
I'll see how it can be corrected.

But note that for blockSize<=3, I don't think arm_biquad_cas_df1_32x64_q31 is going to be efficient. It is made to amortize the overhead of the state copy on a long blockSize.

raysimar, 2020-10-23:
Hi Christophe,

Thanks for the followup.  I am attaching an updated main.c file.

What I had done is prune my original test case a bit too much!  The new one has a block size of 32.  In this case, I get the message “Hardfault occurred!”

I have also debugged it from the entry point and I have found that it gets the fault at a push instruction at the start of main().

Any thoughts?

All the best,
Ray

christophe0606, 2020-10-26:
Hi @raysimar ,

I don't see the updated main.c 
So I have tested with the old one and changed block size to 32.

I don't have any crash.

I assume to help you more we will need to continue this discussion on another channel.

I'll see with our FAE who are already in contact with you.

closed 2020-11-06 by christophe0606