ARM-software/CMSIS_5 #1029: CapsNet support

MiguelCosta94, 2020-10-27:
Do you plan to support Capsule Networks soon?

JonatanAntoni, 2020-11-02:
Hi @MiguelCosta94,

can you be more specific about your request, please?
Which parts of CMSIS do you look at? I suspect it might be about CMSIS-NN.

Cheers,
Jonatan

MiguelCosta94, 2020-11-03:
Hi @JonatanAntoni,

Thanks for the reply. Yes, I was talking about CMSIS-NN. Will it support Capsule Networks in the near future?

Cheers,
Miguel

majianjia, 2020-11-06:
Hi @MiguelCosta94 

CMSIS-NN is more on the backend ops which has nothing to do with manage structure. You May use TFLite micro if they support the layer of your model. Since caps are consist of conv, relu and dense, it shouldnt be a problem.

Alternatively, you can try my work [NNoM](https://github.com/majianjia/nnom) which manage those complex stucture natively and use our high performance CMSIS NN as backend. It should support CapsNet.

felix-johnny, 2021-01-07:
@MiguelCosta94 CMSIS-NN provides optimization for a subset of operators that are supported by TensorFlow Lite Micro. The complete list for TFLu supported operators are in https://github.com/tensorflow/tensorflow/blob/master/tensorflow/lite/micro/all_ops_resolver.cc . 
  The list for CMSIS-NN optimized operators can be found in https://github.com/ARM-software/CMSIS_5/tree/develop/CMSIS/NN#tfl-micro-compliant-apis . If there is an operator supported by TFLu that is not optimized by CMSIS-NN and is a significant cycle consumer in a network, please let us know and we can add support for that. 
 Hope that helps.

closed 2021-01-20 by felix-johnny