ARM-software/CMSIS_5 #1031: Cortex-A15 support and data vector address manipulation - Read align exception.

mrgabich, 2020-10-29:
Hi dear

There is any plan to port the nn kernels to Cortex-A15?
I was trying to port based on Cortex-A libraries and the execution results in "read align" exceptions when interacting with data addressing. I am trying to execute the CIFAR-10 test case and the first convolutional layer reaches this exception.
I just debug the code and found that bug after incrementing the address manipulation variable. For example in arm_convolve_HWC_q7_RGB:
(...)
                        *__SIMD32(pBuffer) = 0x0;
                        *(pBuffer + 2) = 0;
                        pBuffer += 3;
(...)

Further data manipulations have the same alignment problem.

christophe0606, 2020-10-30:
Hi, you may try to replace the deprecated __SIMD32 macros with the functions arm_nn_read_q7x4_ia and similar ones defined in arm_nnsupportfunctions.h

For the writes, if you also have problems with __SIMD32, you can try the similar functions defined in DSP/Include/arm_math_memory.h since I don't see any write functions in arm_nnsupportfunctions.h

felix-johnny, 2020-10-30:
@mrgabich  Thanks for the question. For optimized NN kernels in the Cortex-A class, you can check out the Arm Compute Library. But, to get around the issues that you see as it is in CMSIS-NN, the solution proposed by Christophe is the way forward.

closed 2021-01-07 by felix-johnny