ARM-software/CMSIS_5 #1032: Cortex-M7 ECC bit in CACR

fubo73, 2020-11-02:
Hi,

according to specification ARM Cortex-M7 Processor Technical Reference Manual, revision D, page 3-15, the ECC bit in SCB->CACR is not ECCEN but ECCDIS, with inverted logic. So in core_cm7.h macros:

#define SCB_CACR_ECCEN_Pos                  1U                                            /*!< SCB CACR: ECCEN Position */
#define SCB_CACR_ECCEN_Msk                 (1UL << SCB_CACR_ECCEN_Pos)                    /*!< SCB CACR: ECCEN Mask */

should be changed to

#define SCB_CACR_ECCDIS_Pos                  1U                                            /*!< SCB CACR: ECCDIS Position */
#define SCB_CACR_ECCDIS_Msk                 (1UL << SCB_CACR_ECCDIS_Pos)                    /*!< SCB CACR: ECCDIS Mask */

JonatanAntoni, 2020-11-02:
Hi @fubo73,

thanks for pointing this out.
We'd better just add the new ones in order not to break old code.

Cheers,
Jonatan

JonatanAntoni, 2020-11-03:
Hi @fubo73,

may I ask you to cross check the fix provided in #1033, please?

Thanks,
Jonatan

JonatanAntoni, 2021-06-08:
Closing due to inactivity. Please feel free to reopen if anything left to be discussed.

closed 2021-06-08 by JonatanAntoni