ARM-software/CMSIS_5 #1035: cFFT init for float64 is not declared in arm_math.h

mikeprokoptsov, 2020-11-09:
The function 

arm_status arm_cfft_init_f64(arm_cfft_instance_f64 * S, uint16_t fftLen);

is not declared in arm_math.h in 5_7_0 pack. The same functions for Q15, Q31 and F32 are declared fine.

christophe0606, 2020-11-10:
@mikeprokoptsov You're right. It is missing from the header but built in the library.

This problem is already corrected on the development branch where the header have been reworked a lot.

With 5_7_0, you'll have to add the declaration to one of your header like:

```
extern arm_status arm_cfft_init_f64(
  arm_cfft_instance_f64 * S,
  uint16_t fftLen);
```

mikeprokoptsov, 2020-11-10:
Thank you for the reply. Yes I declared the function in my local copy of header file, I just wanted to let you know about this minor, non-critical issue (less than 1 min to fix locally)
Respectfully
Mike

closed 2020-11-12 by christophe0606