ARM-software/CMSIS_5 #1036: example application for cbuild and GCC

mentaal, 2020-11-10:
Hi there,

I'm trying to get an application built using GCC and CMSIS-build. 

I have tried with an existing application but I've given up on that and decided just to get an example project working to begin with. 
Within Keil, I've taken one of the example blinky projects, in my case SAMV71-XULTRA blinky project which uses RTX5. 
I've modified the project to suit the LEDs on my board, built and run it successfully on my development board. 

I then exported it to a .cprj file.

I then updated the resultant .cprj as follows in order to facilitate compiling with GCC:

1. Added GCC compiler reference: `    <compiler name="GCC" version="9.2.1:9.2.1"/>`
2. Added ldflags: `    <ldflags add="--specs=nosys.specs --specs=nano.specs -lnosys" compiler="GCC"/>`
3. Added some cflags: `    <cflags add="-O0 -ggdb -std=c99" compiler="GCC"/>`

Trying to then build using the command:  `cbuild.sh Blinky.SAMV7_Flash.cprj --toolchain=GCC` yields the following:

```
/../arm-none-eabi/lib/thumb/v7e-m+dp/hard\libnosys.a(sbrk.o): in function `_sbrk':
sbrk.c:(.text._sbrk+0x18): undefined reference to `end'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/temp/rtx5_blinky/Boards/Atmel/SAMV71-XULTRA/Blinky/Flash/Blinky.SAMV7_Flash.mak:111: C:/temp/rtx5_blinky/Boards/Atmel/SAMV71-XULTRA/Blinky/Flash/Blinky.elf] Error 1
make /c/temp/rtx5_blinky/Boards/Atmel/SAMV71-XULTRA/Blinky/Flash/Blinky.SAMV7_Flash.mak failed!
```

I can see that the GCC version of the `.ld` file is referred to in the generated
makefile: `LD_SCRIPT:=C:/temp/rtx5_blinky/Boards/Atmel/SAMV71-XULTRA/Blinky/RTE/Device/ATSAMV71Q21/samv71q21_flash.ld`

So in an attempt to address the previous error, I adjusted the linker script by
adding references to the missing symbols in the bss section:

```
        _sbss = . ;
        __bss_start__ = .; /* my addition here */

        _ebss = . ;
        __bss_end__ = _ebss; /* my addition here */
        end = __bss_end__; /* my addition here */
```

```
/**
 * \file
 *
 * \brief GCC linker script (flash) for ATSAMV71Q21
 *
 * Copyright (c) 2019 Microchip Technology Inc.
 *
 * \license_start
 *
 * \page License
 *
 * SPDX-License-Identifier: Apache-2.0
 *
 * Licensed under the Apache License, Version 2.0 (the "License");
 * you may not use this file except in compliance with the License.
 * You may obtain a copy of the License at
 *
 *   http://www.apache.org/licenses/LICENSE-2.0
 *
 * Unless required by applicable law or agreed to in writing, software
 * distributed under the License is distributed on an "AS IS" BASIS,
 * WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
 * See the License for the specific language governing permissions and
 * limitations under the License.
 *
 * \license_stop
 *
 */

/*------------------------------------------------------------------------------
 *      Linker script for running in internal FLASH on the ATSAMV71Q21
 *----------------------------------------------------------------------------*/

OUTPUT_FORMAT("elf32-littlearm", "elf32-littlearm", "elf32-littlearm")
OUTPUT_ARCH(arm)
SEARCH_DIR(.)

/* Memory Spaces Definitions */
MEMORY
{
    rom (rx)    : ORIGIN = 0x00400000, LENGTH = 0x00200000 /* rom, 2097152K */
    ram (rwx)   : ORIGIN = 0x20400000, LENGTH = 0x00060000 /* ram, 393216K */
}

/* The stack size used by the application. NOTE: you need to adjust according to your application. */
STACK_SIZE = DEFINED(STACK_SIZE) ? STACK_SIZE : DEFINED(__stack_size__) ? __stack_size__ : 0x0400;

/* The heapsize used by the application. NOTE: you need to adjust according to your application. */
HEAP_SIZE = DEFINED(HEAP_SIZE) ? HEAP_SIZE : DEFINED(__heap_size__) ? __heap_size__ : 0x0200;

/* Section Definitions */
SECTIONS
{
    .text :
    {
        . = ALIGN(4);
        _sfixed = .;
        KEEP(*(.vectors .vectors.*))
        *(.text .text.* .gnu.linkonce.t.*)
        *(.glue_7t) *(.glue_7)
        *(.rodata .rodata* .gnu.linkonce.r.*)
        *(.ARM.extab* .gnu.linkonce.armextab.*)

        /* Support C constructors, and C destructors in both user code
           and the C library. This also provides support for C++ code. */
        . = ALIGN(4);
        KEEP(*(.init))
        . = ALIGN(4);
        __preinit_array_start = .;
        KEEP (*(.preinit_array))
        __preinit_array_end = .;

        . = ALIGN(4);
        __init_array_start = .;
        KEEP (*(SORT(.init_array.*)))
        KEEP (*(.init_array))
        __init_array_end = .;

        . = ALIGN(4);
        KEEP (*crtbegin.o(.ctors))
        KEEP (*(EXCLUDE_FILE (*crtend.o) .ctors))
        KEEP (*(SORT(.ctors.*)))
        KEEP (*crtend.o(.ctors))

        . = ALIGN(4);
        KEEP(*(.fini))

        . = ALIGN(4);
        __fini_array_start = .;
        KEEP (*(.fini_array))
        KEEP (*(SORT(.fini_array.*)))
        __fini_array_end = .;

        KEEP (*crtbegin.o(.dtors))
        KEEP (*(EXCLUDE_FILE (*crtend.o) .dtors))
        KEEP (*(SORT(.dtors.*)))
        KEEP (*crtend.o(.dtors))

        . = ALIGN(4);
        _efixed = .;            /* End of text section */
    } > rom

    /* .ARM.exidx is sorted, so has to go in its own output section.  */
    PROVIDE_HIDDEN (__exidx_start = .);
    .ARM.exidx :
    {
      *(.ARM.exidx* .gnu.linkonce.armexidx.*)
    } > rom
    PROVIDE_HIDDEN (__exidx_end = .);

    . = ALIGN(4);
    _etext = .;

    .relocate : AT (_etext)
    {
        . = ALIGN(4);
        _srelocate = .;
        *(.ramfunc .ramfunc.*);
        *(.data .data.*);
        . = ALIGN(4);
        _erelocate = .;
    } > ram

    /* .bss section which is used for uninitialized data */
    .bss (NOLOAD) :
    {
        . = ALIGN(4);
        _sbss = . ;
        __bss_start__ = .; /* my addition here */
        _szero = .;
        *(.bss .bss.*)
        *(COMMON)
        . = ALIGN(4);
        _ebss = . ;
        __bss_end__ = _ebss; /* my addition here */
        _ezero = .;
        end = __bss_end__; /* my addition here */
    } > ram

    /* heap section */
    .heap (NOLOAD):
    {
        . = ALIGN(8);
         _sheap = .;
        . = . + HEAP_SIZE;
        . = ALIGN(8);
        _eheap = .;
    } > ram

    /* stack section */
    .stack (NOLOAD):
    {
        . = ALIGN(8);
        _sstack = .;
        . = . + STACK_SIZE;
        . = ALIGN(8);
        _estack = .;
    } > ram

    . = ALIGN(4);
    _end = . ;
    _ram_end_ = ORIGIN(ram) + LENGTH(ram) - 1 ;
}
```

This allows the project to link successfully.
However, the application just hardfaults when I flash the development board with
the binary. According to Segger's Ozone IDE, it appears to be an instruction access
fault.

[image]

Though, to be honest, I don't understand what's going on based on the IDE's
output as it just seems to be showing PC stepping up through addresses from
0x400_000 (ROM start), as opposed to executing the reset handler.

I'm sure I can do more debug here but I have a feeling I'm going down the wrong
path and am hoping there's an example project I can study as a reference.

As an aside, and I don't really expect you to know this wouldn't necessarily
have anything to do CMSIS but I can see that the system files that
are used in the Keil, arm compiler based build is completely different to the
GCC one. `SystemInit` for example in the GCC flavour has this:

```

    // Keep the default device state after reset
    SystemCoreClock = __SYSTEM_CLOCK;
    return;

```

Whereas the ARM compiler based one: `system_SAMV71.c` does a lot more work in
terms of initialization. Would you have any insight as to why that could be?

Thanks in advance for any information may have here.

brondani, 2020-11-11:
Hi Gregory

The GCC linker and system files that you mentioned are provided by the device vendor, please to refer to them to sort it out and/or to get more examples.
Also note that most SAM packs have been deprecated in favor of new Microchip packs and it seems the migration is still going on.

Best regards
Daniel

mentaal, 2020-11-11:
Will do Daniel thanks.

mentaal, 2020-11-13:
Hi Daniel, 

Ok so I've made some progress on this one.
What I've fixed:
1. Properly set the entry point in the linker script to reference the reset_handler. 
2. Adjusted the startup file to properly reference the SVC_Handler, PendSV_handler and SysTick_Handler. They were called something else for their weakly assigned dummies and so weren't picking up ones from RTX5. This allows the SVC_Handler to be entered correctly instead of the dummy handler.
3. The application still hardfaults when attempting to reference SVC_Count inside the definition of SVC_User in HAL_CM4.sys. The reason for this is that this variable is defined inside the `svc_table` section and this section wasn't referenced within my linker file. Which I guess isn't surprising as it's not a standard section. 
4. I tried adding this section in my linker file. This works in that it no longer hard faults but it's not operating correctly. When I compare against the AC5 build, SVC_Count is a value: 0 in flash memory. 
[image]

Trying it with the GCC build: I end up with a const address to the RAM location  of the variables!
[image]

Would you know I should reference the `svc_table` correctly within the linker file? Or better yet, would you have a reference linker file for an application referencing RTX5? I assume that if RTX5 is supported on GCC, there must be some build environment which loads it somewhere?

The net result of this is that the SVC call doesn't appear to do anything, it enters and then exits, which in my case for a call to `osDelay`, results in no delay at all.

Thanks.

mentaal, 2020-11-14:
I have managed to get a GCC build of the application working with RTX5 and RTOS2.

closed 2020-11-14 by mentaal