ARM-software/CMSIS_5 #1039: Sharp CMSIS-NN performance drop when upgrading to CMSIS 5.7.0

Mjonir, 2020-11-13:
Hello,

I am running an embedded neural network on a nRF52840 platform (Cortex-M4F) generated with the NNoM library with CMSIS-NN backend. Attached below is a comparison of the performance using several releases of CMSIS:
[cmsis_compare.txt](https://github.com/ARM-software/CMSIS_5/files/5537764/cmsis_compare.txt)

I am only changing the revision of CMSIS used when compiling, everything else remains as is. Using the CMSIS 5.4.0 yields the best performance with a prediction time of 720ms. Upgrading up to CMSIS 5.6.0 leads to a slightly higher time of 763ms. However, upgrading from 5.6.0 to 5.7.0 leads to a catastrophic drop in performance putting the prediction time at 6375ms, which is slower than a naive non-optimized implementation. The latest revision from the development branch shows the same degraded performance.

For reference, here is the implemented NN structure. The predicted values seem identical. 
[nn_structure.txt](https://github.com/ARM-software/CMSIS_5/files/5537791/nn_structure.txt)

Would anyone have an explanation for the performance downgrade and possible solutions in how to fix it on the latest revisions?

Thank you in advance.

majianjia, 2020-12-08:
I also met this problem.
The compiler is GCC, CMSIS version is 5.7.0 which downloaded by TensorFlow micro. 
When switching back to 5.6.0. Everything is optimized. I try to step in the CMSIS-NN ops, it runs all the optimized codes (it reached the assembly codes) but the performance is still poor.

felix-johnny, 2021-01-07:
@Mjonir @majianjia .. It is quite likely to be caused by debug compiler options like  -fno-builtin or -ffreestanding . If you are using any of those, please try without it. https://github.com/tensorflow/tensorflow/issues/37361 is a ticket where the issue is discussed.

Mjonir, 2021-01-11:
"-fno-builtin" option was indeed causing the issue. Removing it results in the expected performance increase (5.7.0+ is up 7% from 5.6.0).

Thanks a lot for your advice @felix-johnny !

closed 2021-01-11 by Mjonir

majianjia, 2021-01-12:
Problem solved, thank you very much @felix-johnny

felix-johnny, 2021-01-13:
Glad it resolved the issue!