ARM-software/CMSIS_5 #1042: Is there any plan to support more common drivers(Timer, ADC, DAC, PWM...) in CMSIS-Driver?

windowsair, 2020-11-16:
Would love to have more drivers in CMSIS-Driver such as Timer, ADC, DAC, PWM, etc. Will there be similar plans in the future?
Thanks!

JonatanAntoni, 2020-11-16:
HI @windowsair,

Thanks for raising this.

We do not have immediate plans to add more CMSIS-Driver APIs. Just adding new APIs doesn't bring benefits on its own. We'd rather need to convince device vendors to provide implementations of common HAL APIs. This is another story which needs to be discussed elsewhere.

Just out of curiosity: Do you have specific devices (and/or) vendors in mind when looking for a unified HAL driver interface? Did you already raise this question with these vendors?

Cheers,
Jonatan

windowsair, 2020-11-16:
Thank you for your quick reply！
Different vendors actually give different implementations, and convincing them to use the same API can be a bit difficult.
However the CMSIS project has given me hope for fast migration between different devices.:smiley:

closed 2020-11-16 by windowsair

jankap, 2022-06-14:
Any news on this? We will start a new platform development in our team and the big question is what HAL/Driver to be used. It's likely that we will want to switch between ST and NXP during the development, or support both, due to the recent component shortage issues. The vendor HALs differ a lot, so an official ARM driver interface would be awesome.

Is this API still recommended to be used?

JonatanAntoni, 2022-06-20:
Hi @jankap,

We founded a new open governance project at Linaro for addressing this.
https://www.open-cmsis-cdi.org/

Cheers,
Jonatan