ARM-software/CMSIS_5 #1043: ETM support

metebalci, 2020-11-16:
Is there an ongoing work or a plan to add ETM registers to the codebase like ITM ?

JonatanAntoni, 2020-11-16:
Hi @metebalci,

currently we do not have plans to add ETM registers. Typically, ETM is driven by the debugger and not by the user application.

Do you have a another (specific) use case in mind? 

Cheers,
Jonatan

metebalci, 2020-11-16:
Hi @JonatanAntoni,

I dont have any specific use in mind. I am playing with ETM (and ITM etc.) with an app without a debugger at the moment, and since TPIU, ITM etc. registers are all defined, I think it would make sense to have ETM support in CMSIS as well, at least for the sake of completeness. However, I understand the reason if it is left out.

Best regards,

Mete