ARM-software/CMSIS_5 #1045: Memory requirements for CMSIS DSP functions

AchimSiegfeld, 2020-11-17:
Hi, 
is there any documentation about memory requirements for each DSP function (FFT, FIR, ..., etc.)? I have seen many reports about the number of cycles for these functions, but I could not find any documentation regarding flash requirements. This will help me to decide how big can my project be so that it fits in the internal flash memory.

christophe0606, 2020-11-17:
Hi @AchimSiegfeld ,

There is no documentation about the code size.

But there are lot of compilation flags available to only build and include what you need (like only the FFT tables you need).

AchimSiegfeld, 2020-11-17:
@christophe0606 
thanks for your reply. Where can I find details about these flags? 
I think it would be helpful to have a list of the DSP functions and the required code size + required data size (e.g., coefficients & number of stages of filters).

christophe0606, 2020-11-17:
@AchimSiegfeld  It is not documented. But if you look at arm_cfft_init_f32 you'll see some examples.

You can also look at CommonTables. The files arm_common_tables.c and arm_const_structs.c

By default all those flags are ignored unless you define ARM_DSP_CONFIG_TABLES

closed 2022-02-28 by christophe0606